mavlink/mavlink-camera-manager #17: Create sdp files for stream

patrickelectric, 2022-01-04:
This helps vlc and others programs to run streams without much user configuration

patrickelectric, 2022-01-05:
```
c=IN IP4 0.0.0.0
m=video 5602 RTP/AVP 96
a=rtpmap:96 H264/90000
```

joaoantoniocardoso, 2022-01-07:
For reference, someone did this: https://github.com/webrtc-rs/sdp, which seems more than enough, possibly has more than our needs.

patrickelectric, 2022-09-08:
We can implement UDP h264 for now and rtsp only, everything else we can let for hold, these two should be pretty easy to implement.

ES-Alexander, 2022-10-20:
Relevant to bluerobotics/BlueOS-docker#1195

joaoantoniocardoso, 2022-10-20:
Just writing it down here so I can't lose it in a refactor:

```rust
fn create_sdp_from_pad(pad: gst::Pad) -> Result<gst_webrtc::WebRTCSessionDescription> {
    let caps = pad.caps().context(format!("No caps found in {pad:#?}"))?;
    debug!("Got caps: {caps:#?}");

    let mut sdp_media = gst_sdp::SDPMedia::new();
    gst_sdp::SDPMediaRef::set_media_from_caps(&caps, &mut sdp_media)?;
    debug!("Got SDPMedia: {sdp_media:#?}");

    let mut sdp_message = gst_sdp::SDPMessage::new();
    sdp_message.add_media(sdp_media);
    debug!("Got SDPMessage: {sdp_message:#?}");

    let sdp =
        gst_webrtc::WebRTCSessionDescription::new(gst_webrtc::WebRTCSDPType::Offer, sdp_message);
    debug!("Got SDP: {sdp:#?}");

    Ok(sdp)
}
```

PR #159 (2022-12-21): src: sink: Implement SDP generation for sink interface
* UDP - works with VLC and GStreamer for H264, RAW, and MJPG.
* RTPS - both VLC and GStreamer parse it, but they refuse to play it.

closes #17
closed 2022-12-21 by patrickelectric