mavlink/mavlink-camera-manager #23: settings with multiple streams configured aren't loaded correctly

Williangalvani, 2022-01-11:
See attached log.
[cameramanager.txt](https://github.com/bluerobotics/mavlink-camera-manager/files/7850369/cameramanager.txt)

In master.rs:33, both listed streams start showing as having the same video_source. the code seems to think that the camera moved.

joaoantoniocardoso, 2022-01-12:
Hi @Williangalvani, could you try to reproduce it with the current master?

The reason is that we had noticed that our interpretation of usb bus naming was limited to some extent, and couldn't differentiate devices when using hubs, for example. Since #21, we are using a string to identify each usb device, instead of the `UsbBus` structure, which maybe solves this issue.

Williangalvani, 2022-01-12:
This looks good on master, my bad!

closed 2022-01-12 by Williangalvani