mavlink/mavlink-camera-manager #51: Add command line option to filter some mavlink messages

patrickelectric, 2022-02-14:
The verbose mode with mavlink messages is pretty crazy, it would help if we could add an option with the verbose that we could filter the selected mavlink messages with a regex for names.

patrickelectric, 2022-02-14:
Request from: @Williangalvani

joaoantoniocardoso, 2024-10-18:
We already have the [EnvFilter](https://docs.rs/tracing-subscriber/latest/tracing_subscriber/filter/struct.EnvFilter.html) enabled. Here is a quick ref on  how to use it:

RUST_LOG=<module>=<level>,<module>=<level>,<module>=<level>

since we moved everything to a lib, all our modules lie under the root module `mavlink_camera_manager`: 
Examples:

1. Enable trace for mavlink, keep debug for everything else: `RUST_LOG=debug,mavlink_camera_manager::mavlink=trace`.
2. Shutoff everything from controls, but keep the rest as info: `RUST_LOG=info,mavlink_camera_manager::controls=off`.

For a better fine-grained control one can enable `with_target` from  `fmt::Layer` on our log manager, it shows the target that can be used to filter each log line.

Example: To filter the following line:
```2024-10-18T18:15:51.746587Z DEBUG receiver_loop:connect{address="udpin:0.0.0.0:6666"}: mavlink_camera_manager::mavlink::manager: src/lib/mavlink/manager.rs:233: Connecting...```

One can use:

`RUST_LOG=debug,mavlink_camera_manager::mavlink::manager=off`

PR #444 (2024-10-18): Fix verbosity
To see again those messages:
```
export RUST_LOG=debug,mavlink_camera_manager::controls::onvif=trace,mavlink_camera_manager::mavlink=trace
```

note: a more fine-grained control is also possible if needed

Closes #443
Closes #51
closed 2024-10-18 by joaoantoniocardoso