mavlink/mavlink-camera-manager #57: QGC integration is failing

patrickelectric, 2022-02-24:
Changes on QGC control does not affect v4l controls

joaoantoniocardoso, 2022-05-03:
The part needed for the QGC UI buttons and controls is fixed by #67 and #72, but there are still mavlink messages missing on our side that are sent by QGC, although it seems to be working fine without them.

patrickelectric, 2022-05-03:
Can you inform what kind of messages ? Are messages that are important for us or are just generic vehicle messages that we are ignoring ?

joaoantoniocardoso, 2022-05-03:
@patrickelectric these are the messages not being handled:
- PARAM_EXT_REQUEST_READ ([ #320 ](https://mavlink.io/en/messages/common.html#PARAM_EXT_REQUEST_READ))
- PARAM_EXT_REQUEST_LIST ([ #321 ](https://mavlink.io/en/messages/common.html#PARAM_EXT_REQUEST_LIST))

joaoantoniocardoso, 2022-05-03:
After implementing 320 and 321, there is another being requested:
- MAV_CMD_REQUEST_VIDEO_STREAM_STATUS ([2505](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_VIDEO_STREAM_STATUS) )

joaoantoniocardoso, 2022-08-03:
After #107, we should do a big test to check if anything else is missing. 

From my tests, there is still one message missing, which asks for the value of a control, but QGC ends up asking for the value of all controls when it gets no answer for that message.

patrickelectric, 2022-08-03:
> After implementing 320 and 321, there is another one being requested by QGC:
> 
> * MAV_CMD_REQUEST_VIDEO_STREAM_STATUS ([2505](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_VIDEO_STREAM_STATUS) )

Just check QGC source code

joaoantoniocardoso, 2022-08-03:
> > After implementing 320 and 321, there is another one being requested by QGC:
> > 
> > * MAV_CMD_REQUEST_VIDEO_STREAM_STATUS ([2505](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_VIDEO_STREAM_STATUS) )
> 
> Just check QGC source code

What a great idea, I'll do that right away :1st_place_medal:

PR #107 (2022-08-05): Add missing mavlink message
Closes #42
Closes #57 

- [x] Tested with both UDP and TCP MAVLink endpoint in Raspberry Pi 3, with QGC. 

Helps in both TCP and UDP cases, but none of these changes are necessary in order to make multiple cameras work. This PR is only improving the MAVLink (and therefore QGC) support.
closed 2022-08-05 by patrickelectric