mavlink/mavlink-camera-manager #66: Add support for raspberry camera

patrickelectric, 2022-04-20:
original comment: https://github.com/bluerobotics/BlueOS-docker/issues/991

joaoantoniocardoso, 2022-04-20:
Can we identify the raspberry pi camera with the current implementation?

ES-Alexander, 2022-04-26:
May be relevant:
https://www.raspberrypi.com/documentation/accessories/camera.html?ref=hackernoon.com#network-streaming

ES-Alexander, 2022-04-26:
> Can we identify the raspberry pi camera with the current implementation?

@joaoantoniocardoso `libcamera-hello --list-cameras` seems to detect mine fine, if that's helpful:
```
pi@blueos:~ $ libcamera-hello --list-cameras
[0:08:40.093741943] [1781]  INFO Camera camera manager.cpp:293 libcamera v0.0.0+3384-44d59841
[0:08:40.168778494] [1782]  WARN PI raspberrypi.cpp: 1202 Mismatch between Unicam and CamHelper for embedded data usage!
[0:08:40.169876612] [1782]  INFO PI raspberrypi.cpp: 1313 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
Available cameras
----------------
0 : imx219 [3280x2464] (/base/soc/12c0mux/i2c@1/imx219@10)
    Modes: 'SBGGR10_CSI2P'  : 640x480 1640x1232 1920x1080 3280x2464
           'SBGGR8'  : 640x480 1640x1232 1920x1080 3280x2464
1 : H264 USB Camera: USB Camera [1920x1080] (/base/soc/usb@7e980000/usb1@1-1.2:1.0-32e4:9422)
    Modes: 'MJPEG'  : 320x240 352x288 640x360 640x480 800x600 1280x720 1920x1080 1920x1080
           'YUYV'  : 320x240 352x288 640x360 640x480 640×480 800x600
```

That said, that doesn't seem to detect the H264 output option, and I'm not certain how to ensure it does proper (efficient) hardware encoding like it does with earlier RPiOS versions.

PR #86 (2022-07-18): Add rpicamera support
Closes #66.

Finally, with this PR, Raspberry Pi Camera will be working, but with the following notes:
- Only Legacy Camera mode is supported, to do so, one needs to enable it using `raspi-config`, which can be done in one line: ` if [[ $(raspi-config nonint get_legacy) == 1 ]]; then sudo raspi-config nonint do_legacy 0 && sudo sync && sudo reboot; fi `
- Any Raspberry Pi Camera will be constrained to work with a maximum resolution of 1920 x 1080 px, and/or with a maximum frame rate of 30 FPS.
- rpicam needs to be added into the GStreamer build to have the `rpicamsrc` plu [... cut]
closed 2022-07-18 by patrickelectric