mavlink/mavlink-camera-manager #114: RTSP is not working with QGC

joaoantoniocardoso, 2022-08-17:
The problem is very straightforward:

- We are now advertising via MAVLink the same IP as the user chooses as the stream endpoint when creating the stream
- We are still restricting RTSP streams to have "0.0.0.0" as the stream IP

As consequence, the user will never be able to have the QGC connect to the RTSP stream, only if using an additional redirect stream advertising the _adapter IP_.

Overall this IP thing was quite confusing to me at the beginning, and it might be as well for the end user: while UDP **only** works using the _network IP_ (like `192.168.2.1`), RTSP **only** works when using the _adapter IP_ (like `192.168.2.2`) - will the users know what to choose?

I see two ways to solve this and keep consistency:
A. Just allow the user to choose any IP for RTSP streams, which is what we are doing for any other stream, which isn't a big deal until now.
B. Choose the correct IP for each stream kind, instead of asking for the user to be an expert.

Today the user has to enter the correct IP for each stream kind he is setting up otherwise the stream will never work. Is there any specific reason (that I'm not seeing) for us to ask for the right IP for the users, instead of programmatically choosing the right one for them?

joaoantoniocardoso, 2022-08-29:
Right now, we are choosing to leave the responsibility for the front-end - the front-end should implement their own logic to solve this confusing setup. The BlueOs has implemented a good example (see https://github.com/bluerobotics/BlueOS-docker/pull/1125).

Since #115 this issue is not a bug anymore, so I'm closing it.

closed 2022-08-29 by joaoantoniocardoso

patrickelectric, 2022-08-30:
> bluerobotics/BlueOS-docker#1125

We can provide a mdns for QGC find the IP