mavlink/mavlink-camera-manager #256: Add support to h265 video

patrickelectric, 2023-08-04:


patrickelectric, 2023-08-21:
Based on multiple sources, it appears that h265 support for webrtc will be a major problem.

https://stackoverflow.com/questions/64315827/h265-support-on-webrtc-android/76417043#76417043

Browsers have h265 capabilities (not fully working out of the box) https://caniuse.com/?search=h265.

But h265 support in webrtc is much more problematic, where is not listed among the [mandatory codecs for WebRTC](https://developer.mozilla.org/en-US/docs/Web/Media/Formats/WebRTC_codecs).

patrickelectric, 2023-08-21:
Adding h265 video stream capabilities for QGC, gstreamer and others receives is possible, but WebRTC appears to be a major technology issue on the ecosystem.

joaoantoniocardoso, 2023-08-21:
I'll continue to track [this](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/1494), and comment here if there are any relevant updates.

patrickelectric, 2023-08-21:
To test browser support: https://lf-tk-sg.ibytedtos.com/obj/tcs-client-sg/resources/video_demo_hevc.html

closed 2024-11-03 by joaoantoniocardoso