mavlink/mavlink-camera-manager #257: Add support to cameras with VISCA protocol

patrickelectric, 2023-08-07:
More information here: https://en.wikipedia.org/wiki/VISCA_Protocol

After some discussion, right now, to integrate the VISCA protocol into the project:

Duplicate `video_source_local` to create a `video_source_visca`. This isolates the VISCA-specific implementation while avoiding breaking API and making the implementation of `video_source_local` more complex and harder to maintain.

Given current constraints, refactoring to support multiple protocols per camera source would lead to an API break, which we're currently avoiding. However, introducing a new video_source eases potential future refactoring.

With `video_source_visca` implemented, creating a `VideoSourceViscaType` will handle the specific VISCA camera control types and video sources, much like the `VideoSourceGstType`.

<details>
<summary>A diff that shows the initial steps would be (thanks @joaoantoniocardoso for it):</summary>

```diff
diff --git a/src/server/pages.rs b/src/server/pages.rs
index 13582fe..2120e27 100644
--- a/src/server/pages.rs
+++ b/src/server/pages.rs
@@ -162,6 +162,12 @@ pub async fn v4l() -> Json<Vec<ApiVideoSource>> {
                 formats: redirect.formats(),
                 controls: redirect.controls(),
             },
+            VideoSourceType::Visca(visca) => ApiVideoSource {
+                name: visca.name().clone(),
+                source: visca.source_string().to_string(),
+                formats: visca.formats(),
+                controls: visca.controls(),
+            },
         })
         .collect();
 
diff --git a/src/stream/mod.rs b/src/stream/mod.rs
index 2e9f859..d2f3632 100644
--- a/src/stream/mod.rs
+++ b/src/stream/mod.rs
@@ -221,7 +221,7 @@ impl StreamState {
 
         match &video_and_stream_information.video_source {
             VideoSourceType::Redirect(_) => return Ok(stream), // Do not add any Sink if it's a redirect Pipeline
-            VideoSourceType::Gst(_) | VideoSourceType::Local(_) => (),
+            VideoSourceType::Gst(_) | VideoSourceType::Local(_) | VideoSourceType::Visca(_) => (),
         }
 
         let endpoints = &video_and_stream_information.stream_information.endpoints;
diff --git a/src/stream/pipeline/mod.rs b/src/stream/pipeline/mod.rs
index 2a42609..2a2b1b6 100644
--- a/src/stream/pipeline/mod.rs
+++ b/src/stream/pipeline/mod.rs
@@ -68,7 +68,7 @@ impl Pipeline {
             VideoSourceType::Gst(_) => Pipeline::Fake(FakePipeline {
                 state: pipeline_state,
             }),
-            VideoSourceType::Local(_) => Pipeline::V4l(V4lPipeline {
+            VideoSourceType::Local(_) | VideoSourceType::Visca(_) => Pipeline::V4l(V4lPipeline {
                 state: pipeline_state,
             }),
             VideoSourceType::Redirect(_) => Pipeline::Redirect(RedirectPipeline {
@@ -111,7 +111,7 @@ impl PipelineState {
             VideoSourceType::Gst(_) => {
                 FakePipeline::try_new(pipeline_id, video_and_stream_information)
             }
-            VideoSourceType::Local(_) => {
+            VideoSourceType::Local(_) | VideoSourceType::Visca(_) => {
                 V4lPipeline::try_new(pipeline_id, video_and_stream_information)
             }
             VideoSourceType::Redirect(_) => {
diff --git a/src/video/mod.rs b/src/video/mod.rs
index cbafbb8..5615888 100644
--- a/src/video/mod.rs
+++ b/src/video/mod.rs
@@ -5,3 +5,4 @@ pub mod xml;
 pub mod video_source_gst;
 pub mod video_source_local;
 pub mod video_source_redirect;
+pub mod video_source_visca;
diff --git a/src/video/types.rs b/src/video/types.rs
index f51ea01..462e495 100644
--- a/src/video/types.rs
+++ b/src/video/types.rs
@@ -1,7 +1,7 @@
-use super::video_source::VideoSource;
 use super::video_source_gst::VideoSourceGst;
 use super::video_source_local::VideoSourceLocal;
 use super::video_source_redirect::VideoSourceRedirect;
+use super::{video_source::VideoSource, video_source_visca::VideoSourceVisca};
 use paperclip::actix::Apiv2Schema;
 use serde::{Deserialize, Serialize};
 
@@ -9,6 +9,7 @@ use serde::{Deserialize, Serialize};
 pub enum VideoSourceType {
     Gst(VideoSourceGst),
     Local(VideoSourceLocal),
+    Visca(VideoSourceVisca),
     Redirect(VideoSourceRedirect),
 }
 
@@ -97,6 +98,7 @@ impl VideoSourceType {
             VideoSourceType::Local(local) => local,
             VideoSourceType::Gst(gst) => gst,
             VideoSourceType::Redirect(redirect) => redirect,
+            VideoSourceType::Visca(visca) => visca,
         }
     }
 }
diff --git a/src/video/video_source_visca.rs b/src/video/video_source_visca.rs
new file mode 100644
index 0000000..c2a4920
--- /dev/null
+++ b/src/video/video_source_visca.rs
@@ -0,0 +1,80 @@
+use std::collections::HashMap;
+use std::sync::{Arc, Mutex};
+
+use super::types::*;
+use super::video_source::{VideoSource, VideoSourceAvailable};
+use paperclip::actix::Apiv2Schema;
+use serde::{Deserialize, Serialize};
+
+use lazy_static::lazy_static;
+
+lazy_static! {
+    static ref VIDEO_FORMATS: Arc<Mutex<HashMap<String, Vec<Format>>>> = Default::default();
+}
+
+//TODO: Move to types
+#[derive(Apiv2Schema, Clone, Debug, PartialEq, Eq, Deserialize, Serialize)]
+pub enum VideoSourceViscaType {
+    Ethernet(String),
+    Rs232(String),
+}
+
+#[derive(Apiv2Schema, Clone, Debug, Deserialize, PartialEq, Eq, Serialize)]
+pub struct VideoSourceVisca {
+    pub name: String,
+    pub device_path: String,
+    #[serde(rename = "type")]
+    pub typ: VideoSourceViscaType,
+}
+
+impl VideoSourceViscaType {}
+
+impl VideoSourceVisca {}
+
+impl VideoSource for VideoSourceVisca {
+    fn name(&self) -> &String {
+        &self.name
+    }
+
+    fn source_string(&self) -> &str {
+        &self.device_path
+    }
+
+    fn formats(&self) -> Vec<Format> {
+        todo!()
+    }
+
+    fn set_control_by_name(&self, control_name: &str, value: i64) -> std::io::Result<()> {
+        todo!()
+    }
+
+    fn set_control_by_id(&self, control_id: u64, value: i64) -> std::io::Result<()> {
+        todo!()
+    }
+
+    fn control_value_by_name(&self, control_name: &str) -> std::io::Result<i64> {
+        todo!()
+    }
+
+    fn control_value_by_id(&self, control_id: u64) -> std::io::Result<i64> {
+        todo!()
+    }
+
+    fn controls(&self) -> Vec<Control> {
+        todo!()
+    }
+
+    fn is_valid(&self) -> bool {
+        !self.device_path.is_empty()
+    }
+
+    fn is_shareable(&self) -> bool {
+        false
+    }
+}
+
+impl VideoSourceAvailable for VideoSourceVisca {
+    fn cameras_available() -> Vec<VideoSourceType> {
+        todo!()
+    }
+}
```
</details>

ShivangRawat30, 2023-08-07:
Hey I am currently learning rust, as it is a good first issue can you please assign this to me

patrickelectric, 2023-08-08:
Hi @ShivangRawat30, thanks! That would be great! Do you have a VISCA camera with you ?
You can also check the others issues that I just tagged as good first issue. Let me know if you want to do any of them.