mavlink/mavlink-camera-manager #260: Resource temporarily unavailable when configure stream

amerghazal7, 2023-08-10:
Hello there, 
After installing and building I tried to create a stream from my USB camera connected to the Jetson Orin, but nothing happens in the UI and I'm getting the error below in the console:

```
camera.rs:816: Error receiving a message as 1:100. Reason: Io(
    Os {
        code: 11,
        kind: WouldBlock,
        message: "Resource temporarily unavailable",
    },
). Camera: MavlinkCameraInformation {
    component: MavlinkCameraComponent {
        system_id: 1,
        component_id: 100,
        stream_id: 1,
        vendor_name: "Logitech BRIO",
        model_name: "/dev/video0 - Logitech BRIO",
        firmware_version: 0,
        resolution_h: 1080,
        resolution_v: 1920,
        framerate: 30.0,
        bitrate: 5000,
        rotation: 0,
        hfov: 90,
        thermal: false,
    },
    mavlink_connection_string: "tcpout:127.0.0.1:5760",
    mavlink_stream_type: VIDEO_STREAM_TYPE_RTPUDP,
    video_stream_uri: Url {
        scheme: "udp",
        cannot_be_a_base: false,
        username: "",
        password: None,
        host: Some(
            Domain(
                "0.0.0.0",
            ),
        ),
        port: Some(
            5600,
        ),
        path: "/logi",
        query: None,
        fragment: None,
    },
    video_source_type: Local(
        VideoSourceLocal {
            name: "Logitech BRIO",
            device_path: "/dev/video0",
            typ: Unknown(
                "usb-3610000.xhci-3.4",
            ),
        },
    ),
}

```

**NOTE** : there's no other apps or processes using the camera, and I'm using mavlink-router to provide a server for the mavlink communication.

Any idea how to solve that?

patrickelectric, 2023-08-10:
It appears that is failing to communicate with the mavlink server, are you running mavlink-router as tcpin ? Be sure to start the router before the camera manager.

amerghazal7, 2023-08-10:
I'm running the default config, this is the output for mavlink-server

[image]

amerghazal7, 2023-08-10:
and the config is as below
```

[General]
   TcpServerPort=5760
   ReportStats=false
   MavlinkDialect=common

[UdpEndpoint local]
   Mode=normal
   Address=127.0.0.1
   Port=14550

```

joaoantoniocardoso, 2023-08-10:
Hi @amerghazal7, I don't think the mavlink is, in fact, the problem, that message appears also when it's working, it is a resource-sharing error, but it usually recovers from it, and is able to send the messages. I'm now refactoring the mavlink part of the code to get rid of it.

For us to hunt the problem, can you:
1. Run `export GST_DEBUG=3` before calling mavlink-camera-manager
2. Call mavlink-camera-manager with `--verbose` as argument
3. Share a full log file running

Thank you, hope we can help

amerghazal7, 2023-08-10:
@joaoantoniocardoso First of all, thank you very much for your quick response and I'm more than happy to help.
Here's the complete terminal output after running the above commands
```

2023-08-10T13:15:16.478512Z  INFO main ThreadId(01) src/logger/manager.rs:82: mavlink-camera-manager, version: 0.2.4-7b4f7eb, build date: 2023-08-10
2023-08-10T13:15:16.479181Z  INFO main ThreadId(01) src/logger/manager.rs:89: Starting at 2023-08-10T17:15:16
2023-08-10T13:15:16.501818Z  WARN             main ThreadId(01) src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"usb-3610000.xhci-3.4\""
2023-08-10T13:15:16.504606Z  WARN             main ThreadId(01) src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"usb-3610000.xhci-3.4\""
2023-08-10T13:15:16.507341Z  WARN             main ThreadId(01) src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"usb-3610000.xhci-3.4\""
2023-08-10T13:15:16.509670Z  WARN             main ThreadId(01) src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"usb-3610000.xhci-3.4\""
2023-08-10T13:15:16.511258Z  INFO             main ThreadId(01) /home/microorin1/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/builder.rs:200: starting 8 workers
2023-08-10T13:15:16.511417Z  INFO             main ThreadId(01) /home/microorin1/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/server.rs:196: Actix runtime found; starting in Actix runtime
2023-08-10T13:15:37.151741Z  INFO actix-rt|system:0|arbiter:0 ThreadId(21) /home/microorin1/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-web-4.3.1/src/middleware/logger.rs:421: 10.20.0.183 "GET / HTTP/1.1" 200 21058 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0" 0.001794
2023-08-10T13:15:37.317711Z  INFO actix-rt|system:0|arbiter:0 ThreadId(21) /home/microorin1/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-web-4.3.1/src/middleware/logger.rs:421: 10.20.0.183 "GET /vue.js HTTP/1.1" 200 542311 "http://10.20.0.202:6020/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0" 0.035788
2023-08-10T13:15:37.458536Z  WARN actix-rt|system:0|arbiter:0 ThreadId(21) HTTP request{http.method=GET http.route=/v4l http.flavor=1.1 http.scheme=http http.host=10.20.0.202:6020 http.client_ip=10.20.0.183 http.user_agent=Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0 http.target=/v4l otel.name=HTTP GET /v4l otel.kind="server" request_id=9f18397f-ff97-4119-acb2-8c715f0c55a2}: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"usb-3610000.xhci-3.4\""
2023-08-10T13:15:37.464275Z  WARN actix-rt|system:0|arbiter:0 ThreadId(21) HTTP request{http.method=GET http.route=/v4l http.flavor=1.1 http.scheme=http http.host=10.20.0.202:6020 http.client_ip=10.20.0.183 http.user_agent=Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0 http.target=/v4l otel.name=HTTP GET /v4l otel.kind="server" request_id=9f18397f-ff97-4119-acb2-8c715f0c55a2}: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"usb-3610000.xhci-3.4\""
2023-08-10T13:15:37.469719Z  WARN actix-rt|system:0|arbiter:0 ThreadId(21) HTTP request{http.method=GET http.route=/v4l http.flavor=1.1 http.scheme=http http.host=10.20.0.202:6020 http.client_ip=10.20.0.183 http.user_agent=Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0 http.target=/v4l otel.name=HTTP GET /v4l otel.kind="server" request_id=9f18397f-ff97-4119-acb2-8c715f0c55a2}: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"usb-3610000.xhci-3.4\""
2023-08-10T13:15:37.474802Z  WARN actix-rt|system:0|arbiter:0 ThreadId(21) HTTP request{http.method=GET http.route=/v4l http.flavor=1.1 http.scheme=http http.host=10.20.0.202:6020 http.client_ip=10.20.0.183 http.user_agent=Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0 http.target=/v4l otel.name=HTTP GET /v4l otel.kind="server" request_id=9f18397f-ff97-4119-acb2-8c715f0c55a2}: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"usb-3610000.xhci-3.4\""
2023-08-10T13:15:37.495359Z  INFO actix-rt|system:0|arbiter:0 ThreadId(21) /home/microorin1/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-web-4.3.1/src/middleware/logger.rs:421: 10.20.0.183 "GET /v4l HTTP/1.1" 200 24386 "http://10.20.0.202:6020/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0" 0.047641
2023-08-10T13:15:37.526617Z  INFO actix-rt|system:0|arbiter:0 ThreadId(21) /home/microorin1/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-web-4.3.1/src/middleware/logger.rs:421: 10.20.0.183 "GET /streams HTTP/1.1" 200 2 "http://10.20.0.202:6020/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/116.0" 0.001403
[... cut ...]
        cannot_be_a_base: false,
        username: "",
        password: None,
        host: Some(
            Domain(
                "0.0.0.0",
            ),
        ),
        port: Some(
            5600,
        ),
        path: "",
        query: None,
        fragment: None,
    },
    video_source_type: Local(
        VideoSourceLocal {
            name: "Logitech BRIO",
            device_path: "/dev/video0",
            typ: Unknown(
                "usb-3610000.xhci-3.4",
            ),
        },
    ),
}

```

amerghazal7, 2023-08-10:
Now after building the new merge, I can see the stream configuration as below:
[image]

but the error still coming up in the terminal, and in the WebRTC client I can't Add a session, see image below:
[image]

joaoantoniocardoso, 2023-08-10:
Alright, now it's a software limitation: our WebRTC feature only supports H264 cameras. To add support for JPEG, we'd have to decode it from the JPEG and reencode it as VP9 or h264, which would add some latency and CPU usage.

Regardless of WebRTC, you can still receive it through UDP or RTSP.

amerghazal7, 2023-08-10:
Thank you very much for your feedback, I'm going to play arround to write a C++ consumer for the stream to see if I'm satisfied with the latency (I would appreciate some guidance to!).
Regarding adding JPEG support to your WebRTC I would be more than happy to know how and contribute.

joaoantoniocardoso, 2023-08-10:
> Thank you very much for your feedback, I'm going to play arround to write a C++ consumer for the stream to see if I'm satisfied with the latency (I would appreciate some guidance too!).

Nice! Well, to receive you can start testing with the following pipeline right from your console:

```
gst-launch-1.0 -vc udpsrc address=0.0.0.0 port=5600 close-socket=false auto-multicast=true \
    ! application/x-rtp,payload=96 ! rtpjpegdepay ! jpegdec \
    ! videoconvert ! fpsdisplaysink sync=false
```

If that works, you can start by hacking [this example](https://gstreamer.freedesktop.org/documentation/tutorials/basic/hello-world.html?gi-language=c#hello-world).

---

>Regarding adding JPEG support to your WebRTC I would be more than happy to know how and contribute.

Awesome, I'll prepare a guide and link it here later.

patrickelectric, 2023-08-10:
You can also use C++ to receive the stream, you can check this [minimal example](https://gist.github.com/patrickelectric/5dca1cb7cef4ffa7fbb6fb70dd9f9edc).

It's also possible to write a C++/Qt application that uses [this QML code](https://gist.github.com/patrickelectric/0c23731db2e2a05ca38304a6c69e1af2), or just the QML with [QHot](https://github.com/patrickelectric/qhot).

Or [Python](https://gist.github.com/patrickelectric/443645bb0fd6e71b34c504d20d475d5a).

patrickelectric, 2023-08-10:
Closing this for now since the underlying problem was detected.

closed 2023-08-10 by patrickelectric