mavlink/mavlink-camera-manager #268: Support for raspberry pi camera

londi, 2023-08-14:
Hi
I'm not sure if it's a bug or if the raspberry pi connected via csi is not compatible.
I use a raspberry pi 4 with 8GB Ram and raspbian 32bit.

I do not see any settings about fps and be not able to start a stream.
[image]

(the camera is working)

londi, 2023-08-14:
update: same issue with raspberry pi and an ordinary usb camera

patrickelectric, 2023-08-14:
it should work with usb cameras without problems, what kind of behaviour are you seeing ?

londi, 2023-08-14:
I have started mavlink-camera-manager as follow:
`./mavlink-camera-manager-armv7 --mavlink=serial:/dev/ttyACM0:57600`

I can configure the stream of a usb camera:
[image]

But I can not create a Session on the webrtc page:

[image]

And on Qgroundcontrol is no image visible. 

The usb camera led is on as soon as I start the stream (configure stream button).

Have you an idea what the problem could be? :)

joaoantoniocardoso, 2023-08-17:
Hi @londi, can you try to enable camera legacy mode? You can do it by running the following on a terminal inside your Raspberry Pi:

```bash
sudo raspi-config nonint do_legacy 0
```

(note: `0` is to **enable**, `1` to **disable** the legacy mode)

Then try to locate a new device `mmal`:

[image]

Hope it helps, thanks

joaoantoniocardoso, 2024-04-27:
I'm closing it for now, feel free to comment or create other issues as you wish.

closed 2024-04-27 by joaoantoniocardoso