mavlink/mavlink-camera-manager #269: How to create a mavlink camera from an RTSP IP camera stream?

joaoantoniocardoso, 2023-08-17:
>Hello,
>I don't know if this is the right place to ask my question but it's not an actual issue so I'll ask it here.
>How do I create a redirect stream from a rtsp IP camera stream?
>I created a gstreamer pipline to receive it and it's working fine, but how do I create a >MAVLINK stream from it? the device in the UI for redirect source is a fixed text, where do >I specify my rtsp link an create a redirect out of it?

_Originally posted by @aihamshikhomicro in https://github.com/mavlink/mavlink-camera-manager/issues/213#issuecomment-1682032408_

joaoantoniocardoso, 2023-08-17:
Hi @aihamshikhomicro, thanks for reaching out, we indeed lack documentation, but it is doable:

## How to create a Mavlink camera from an RTSP IP camera stream?

Let's say we want to receive an RTSP stream from an IP camera on QGC. 

For this example, we are going to consider that the QGC is running on a computer at the user's computer (`localhost`), where its local network IP is `192.168.0.10`, and both Mavlink server and MCM are running on a second computer, the `192.68.0.87`, and the camera is also on the same local network, at `192.168.0.201`.

For that, we'd need:

1. A Mavlink UDP server to connect to QGC (QGroundControl), say, `192.168.0.10:14550`
2. A Mavlink TCP server to connect to MCM (Mavlink Camera Manager), `127.0.0.1:5777`
3. The RTSP address from your IP camera, `rtsp://192.168.0.201:554/video/0`
4. A reliable network connection between all these parts.

Now, the setup should be quite straightforward:

1. Run mavlink-camera-manager passing the Mavlink TCP server endpoint:
```bash
mavlink-camera-manager --mavlink tcpout:127.0.0.1:5777 --verbose
```
2. From the MCM developer interface (`192.168.0.87:6020`), create a Redirect stream using the RTSP address from the camera, let's say, `rtsp://192.168.0.201:554/video/0`. Make sure to leave `Disable Mavlink` **unchecked**:

[image]

3. Finally, open QGC, it should be able to receive the IP stream.

---

If anything different happens, check on the console, and upload the logs here on this issue.

In case your network or setup is different, let us know how it is so I can adapt the guide if you need it,

Hope it helps, thanks

hamishwillee, 2023-08-17:
FWIW sounds like a good topic to add to a docs library.

patrickelectric, 2023-08-21:
I would suggest to just upgrade the interface to be more user friendly :laughing: