mavlink/mavlink-camera-manager #270: Error during build: "failed to run custom build command for `mavlink-camera-manager v0.2.4`"  - Building from source on NXP NAVQPLUS (4x Arm Cortex A53)

steamfoundry, 2023-08-21:
Sorry I can't be more helpful.   Rust is outside my current expertise

### Building steps
- Followed steps to build in README.md
- checked out release T3.12.1
- Received error
- Re-ran with RUST_BACKTRACE=1 set as recommended for more information
- Re-re-ran with RUST_BACKTRACE=full set.

### Board details

- Board - [NXP NavQPlus ](https://nxp.gitbook.io/navqplus/)
- Processor -NXP i.MX 8M Plus (4x Arm Cortex A53 cores_
- OS: Ubuntu 22.04 on Yocto Linux 5.15 kernel - [Build 22.04.3 from Rudis Labs](https://github.com/rudislabs/navqplus-create3-images/releases/tag/v22.04.3)

### Run with RUST_BACKTRACE=1
```
user@NAVQP:~/src/mavlink-camera-manager$ bash -c "RUST_BACKTRACE=1 cargo build" 
warning: unused manifest key: package.path
   Compiling mavlink-camera-manager v0.2.4 (/home/user/src/mavlink-camera-manager)
error: failed to run custom build command for `mavlink-camera-manager v0.2.4 (/home/user/src/mavlink-camera-manager)`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_DEV_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.

Caused by:
  process didn't exit successfully: `/home/user/src/mavlink-camera-manager/target/debug/build/mavlink-camera-manager-70378a934bdfa2ae/build-script-build` (exit status: 101)
  --- stdout
  cargo:rustc-env=VERGEN_BUILD_DATE=2023-08-21
  cargo:rustc-env=VERGEN_BUILD_SEMVER=0.2.4
  cargo:rustc-env=VERGEN_GIT_BRANCH=detached HEAD
  cargo:rustc-env=VERGEN_GIT_COMMIT_AUTHOR_EMAIL=joao.maker@gmail.com
  cargo:rustc-env=VERGEN_GIT_COMMIT_AUTHOR_NAME=João Antônio Cardoso
  cargo:rustc-env=VERGEN_GIT_COMMIT_COUNT=750
  cargo:rustc-env=VERGEN_GIT_COMMIT_MESSAGE=src: stream, rtsp: Remove stream if the path is already in use

  cargo:rustc-env=VERGEN_GIT_COMMIT_TIMESTAMP=2023-08-10T21:37:06Z
  cargo:rustc-env=VERGEN_GIT_SEMVER=0.2.4
  cargo:rustc-env=VERGEN_GIT_SHA_SHORT=5e2a1f8
  cargo:rerun-if-changed=/home/user/src/mavlink-camera-manager/.git/HEAD
  cargo:rerun-if-changed=/home/user/src/mavlink-camera-manager/.git/HEAD
  cargo:rerun-if-changed=src/signalling_protocol.rs
  cargo:rerun-if-changed=./src/stream/webrtc/frontend/index.html
  cargo:rerun-if-changed=./src/stream/webrtc/frontend/yarn.json
  cargo:rerun-if-changed=./src/stream/webrtc/frontend/src

  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', build.rs:97:10
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/eb26296b556cef10fb713a38f3d16b9886080f26/library/std/src/panicking.rs:593:5
     1: core::panicking::panic_fmt
               at /rustc/eb26296b556cef10fb713a38f3d16b9886080f26/library/core/src/panicking.rs:67:14
     2: core::result::unwrap_failed
               at /rustc/eb26296b556cef10fb713a38f3d16b9886080f26/library/core/src/result.rs:1651:5
     3: core::result::Result<T,E>::unwrap
     4: build_script_build::build_with_yarn
     5: build_script_build::main
     6: core::ops::function::FnOnce::call_once
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

### Run with RUST_BACKTRACE=full
```

user@NAVQP:~/src/mavlink-camera-manager$ bash -c "RUST_BACKTRACE=full cargo build" 
warning: unused manifest key: package.path
   Compiling mavlink-camera-manager v0.2.4 (/home/user/src/mavlink-camera-manager)
error: failed to run custom build command for `mavlink-camera-manager v0.2.4 (/home/user/src/mavlink-camera-manager)`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_DEV_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.

Caused by:
[... 65 lines cut ...]
                                 at /rustc/eb26296b556cef10fb713a38f3d16b9886080f26/library/std/src/panicking.rs:500:40
    25:     0xaaaac71e0a88 - std::panicking::try::h4ca46afb085bd0cf
                                 at /rustc/eb26296b556cef10fb713a38f3d16b9886080f26/library/std/src/panicking.rs:464:19
    26:     0xaaaac71e0a88 - std::panic::catch_unwind::h3a523898b68bfb9e
                                 at /rustc/eb26296b556cef10fb713a38f3d16b9886080f26/library/std/src/panic.rs:142:14
    27:     0xaaaac71e0a88 - std::rt::lang_start_internal::{{closure}}::h32ffde1fa2068b4d
                                 at /rustc/eb26296b556cef10fb713a38f3d16b9886080f26/library/std/src/rt.rs:148:48
    28:     0xaaaac71e0a88 - std::panicking::try::do_call::hf293938f127230a2
                                 at /rustc/eb26296b556cef10fb713a38f3d16b9886080f26/library/std/src/panicking.rs:500:40
    29:     0xaaaac71e0a88 - std::panicking::try::h2995b7824979d7b9
                                 at /rustc/eb26296b556cef10fb713a38f3d16b9886080f26/library/std/src/panicking.rs:464:19
    30:     0xaaaac71e0a88 - std::panic::catch_unwind::h16e1cf0a6dcfc682
                                 at /rustc/eb26296b556cef10fb713a38f3d16b9886080f26/library/std/src/panic.rs:142:14
    31:     0xaaaac71e0a88 - std::rt::lang_start_internal::h0f571199619736f5
                                 at /rustc/eb26296b556cef10fb713a38f3d16b9886080f26/library/std/src/rt.rs:148:20
    32:     0xaaaac6bb8594 - std::rt::lang_start::h5c52f0fbae708ef3
    33:     0xaaaac6b93530 - main
    34:     0xffffa34073fc - __libc_start_call_main
                                 at ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
    35:     0xffffa34074cc - __libc_start_main_impl
                                 at ./csu/../csu/libc-start.c:392:3
    36:     0xaaaac6b926f0 - _start
    37:                0x0 - <unknown>
user@NAVQP:~/src/mavlink-camera-manager$ 
```

joaoantoniocardoso, 2023-08-21:
Hi @steamfoundry, the problem is probably my fault as I forgot to mention the Yarn requirement and installation (issue https://github.com/mavlink/mavlink-camera-manager/issues/247).

Basically what it means is that NodeJS and Yarn are used to build the frontend part of the application. I'm not sure exactly for your Ubuntu version, but this is how I do it for a Debian:bullseye-slim docker:

```Bash
# Install NodeJs and Yarn
RUN curl -fsSL https://deb.nodesource.com/setup_19.x | bash - &&\
    curl -sS https://dl.yarnpkg.com/debian/pubkey.gpg | apt-key add &&\
    echo "deb https://dl.yarnpkg.com/debian/ stable main" | tee /etc/apt/sources.list.d/yarn.list &&\
    apt-get update && apt-get install --assume-yes --no-install-recommends \
        nodejs \
        yarn
```

Hope it helps

steamfoundry, 2023-08-21:
Thanks.  I'll try that

joaoantoniocardoso, 2023-08-21:
@steamfoundry, I've just created a PR adding the following instructions (it's for Ubuntu):

```Bash
curl -fsSL https://deb.nodesource.com/setup_19.x \
  | sudo -E bash -
curl -sS https://dl.yarnpkg.com/debian/pubkey.gpg \
  | sudo apt-key add
echo "deb https://dl.yarnpkg.com/debian/ stable main" \
  | sudo tee /etc/apt/sources.list.d/yarn.list
apt-get update && apt-get install --assume-yes --no-install-recommends \
  nodejs \
  yarn
```

Can you confirm it works?

steamfoundry, 2023-08-21:
The instructions worked for adding them.  I ran them as root, not as a user though.  I'll have to test that separately but they look correct.
Now I'll test the build again and see if that was the only issue.

steamfoundry, 2023-08-21:
```
user@NAVQP:~/src/mavlink-camera-manager$ cargo build
warning: unused manifest key: package.path
   Compiling mavlink-camera-manager v0.2.4 (/home/user/src/mavlink-camera-manager)
    Finished dev [unoptimized + debuginfo] target(s) in 6m 07s
warning: the following packages contain code that will be rejected by a future version of Rust: nom v5.1.2
note: to see what the problems were, use the option `--future-incompat-report`, or run `cargo report future-incompatibilities --id 1`
user@NAVQP:~/src/mavlink-camera-manager$ 
```

And on first run we get ```

./mavlink-camera-manager 
2023-08-21T23:09:23.492666Z  INFO main ThreadId(01) src/logger/manager.rs:82: mavlink-camera-manager, version: 0.2.4-5e2a1f8, build date: 2023-08-21
2023-08-21T23:09:23.496208Z  INFO main ThreadId(01) src/logger/manager.rs:89: Starting at 2023-08-21T23:09:23
2023-08-21T23:09:23.499147Z ERROR main ThreadId(01) src/settings/manager.rs:121: Failed to load settings file "/home/user/.config/mavlink-camera-manager/settings.json". Reason: No such file or directory (os error 2)
2023-08-21T23:09:23.499514Z  INFO main ThreadId(01) src/settings/manager.rs:107: The settings file "/home/user/.config/mavlink-camera-manager/settings.json" will be backed-up as "/home/user/.config/mavlink-camera-manager/settings.json.bak", and a new (empty) settings file will be created in its place.
2023-08-21T23:09:23.499725Z ERROR main ThreadId(01) src/settings/manager.rs:110: Failed to create backup file "/home/user/.config/mavlink-camera-manager/settings.json.bak". Reason: Os {
    code: 2,
    kind: NotFound,
    message: "No such file or directory",
}
2023-08-21T23:09:23.587959Z  WARN             main ThreadId(01) src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:cap_devic\""
2023-08-21T23:09:23.607191Z  WARN             main ThreadId(01) src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:m2m_devic\""
2023-08-21T23:09:23.613098Z ERROR             main ThreadId(01) src/video/video_source_local.rs:623: Faield to get device "/dev/video1": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-08-21T23:09:23.613732Z ERROR             main ThreadId(01) src/video/video_source_local.rs:623: Faield to get device "/dev/video0": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-08-21T23:09:23.616501Z  INFO             main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/builder.rs:200: starting 4 workers
2023-08-21T23:09:23.616874Z  INFO             main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/server.rs:196: Actix runtime found; starting in Actix runtime
^C2023-08-21T23:09:36.182640Z  INFO             main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/server.rs:319: SIGINT received; starting forced shutdown
2023-08-21T23:09:36.183190Z  INFO actix-rt|system:0|arbiter:0 ThreadId(13) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/worker.rs:591: shutting down idle worker
2023-08-21T23:09:36.183190Z  INFO actix-rt|system:0|arbiter:1 ThreadId(14) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/worker.rs:591: shutting down idle worker
2023-08-21T23:09:36.183386Z  INFO       actix-server acceptor ThreadId(17) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/accept.rs:143: accept thread stopped
2023-08-21T23:09:36.183196Z  INFO actix-rt|system:0|arbiter:2 ThreadId(15) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/worker.rs:591: shutting down idle worker
2023-08-21T23:09:36.184605Z  INFO actix-rt|system:0|arbiter:3 ThreadId(16) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/worker.rs:591: shutting down idle worker

I'll create the config folder and see how it goes from there.  

```

closed 2023-08-21 by steamfoundry

joaoantoniocardoso, 2023-08-22:
That's a logical bug, I'll fix it asap. You should be fine creating that directory (`/home/user/.config/mavlink-camera-manager`) for now, thanks!

AlexKlimaj, 2023-11-16:
Ran into this same issue. Build gets further after installing yarn but still fails. Creating the src/stream/webrtc/frontend/dist folder allows the build to finish. 

```
jetson@tegra-ubuntu:~/mavlink-camera-manager$ cargo build
warning: unused manifest key: package.path
   Compiling mavlink-camera-manager v0.2.4 (/home/jetson/mavlink-camera-manager)
00h00m00s 0/0: :     Building [=======================> ] 555/556: mavlink-camera-manager(bin)                                                                                   error: proc macro panicked
  --> src/server/pages.rs:84:31
   |
84 | static WEBRTC_DIST: Dir<'_> = include_dir!("src/stream/webrtc/frontend/dist");
   |                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = help: message: "src/stream/webrtc/frontend/dist" is not a directory

error: could not compile `mavlink-camera-manager` (bin "mavlink-camera-manager") due to previous error
```

tranhien1612, 2024-09-20:
@joaoantoniocardoso 
I used source code t3.16.0 and when building it had an error, I used additional commands below but the error still occurred.
```
curl -fsSL https://deb.nodesource.com/gpgkey/nodesource-repo.gpg.key \
  | sudo gpg --dearmor -o /usr/share/keyrings/nodesource.gpg &&\
echo "deb [signed-by=/usr/share/keyrings/nodesource.gpg] https://deb.nodesource.com/node_20.x nodistro main" \
  | sudo tee /etc/apt/sources.list.d/nodesource.list &&\
curl -fsSL https://dl.yarnpkg.com/debian/pubkey.gpg \
  | sudo gpg --dearmor -o /usr/share/keyrings/yarnkey.gpg &&\
echo "deb [signed-by=/usr/share/keyrings/yarnkey.gpg] https://dl.yarnpkg.com/debian stable main" \
  | sudo tee /etc/apt/sources.list.d/yarn.list &&\
sudo apt-get update -y &&\
  sudo apt-get install -y --no-install-recommends \
  nodejs \
  yarn
```
or
```
curl -fsSL https://deb.nodesource.com/setup_19.x \
  | sudo -E bash -
curl -sS https://dl.yarnpkg.com/debian/pubkey.gpg \
  | sudo apt-key add
echo "deb https://dl.yarnpkg.com/debian/ stable main" \
  | sudo tee /etc/apt/sources.list.d/yarn.list
apt-get update && apt-get install --assume-yes --no-install-recommends \
  nodejs \
  yarn
```
[image]

joaoantoniocardoso, 2024-09-20:
Hi, @tranhien1612,

Unfortunately, the build requires the .git folder, so we don't support building it from the released zip/tar (the compressed source code files).

The same can be achieved with the following command:

```bash
git clone --branch t3.16.0 --depth 1 https://github.com/mavlink/mavlink-camera-manager.git
```

tranhien1612, 2024-09-21:
Hi @joaoantoniocardoso ,
Initially I downloaded the source code according to your instructions:
```
git clone https://github.com/mavlink/mavlink-camera-manager.git &&\
cd mavlink-camera-manager
```
, but when building the above error still occurred, so then I tried downloading the t3.16.0 source and building it but the error still occurred.

patrickelectric, 2024-09-21:
Don't use `sudo cardo build`.

```
cd /tmp && git clone --branch t3.16.0 --depth 1 https://github.com/mavlink/mavlink-camera-manager.git && cd mavlink-camera-manager && cargo build
```
Run the following command and report the error if occurs.

tranhien1612, 2024-09-23:
@patrickelectric Thanks for your advice, I tried it but it still gives the same error.
[image]

patrickelectric, 2024-09-23:
It appears that `bun` is not installed in your system. 
https://bun.sh/docs/installation

tranhien1612, 2024-09-23:
@patrickelectric I installed ```bun``` according to the instructions but the error still occurs.

[image]

patrickelectric, 2024-09-23:
Can you run `bun --version` in your terminal ?

tranhien1612, 2024-09-23:
@patrickelectric I had to run another command ``` source /home/maj/.bashrc``` and when I finished running it, the error was fixed.

[image]

patrickelectric, 2024-09-23:
Are you able to compile camera manager now ?

tranhien1612, 2024-09-23:
@patrickelectric Yes, it has created the executable file. Thank you for your suggestion.

I am trying to display image from ```WebCam``` or ```Camera: Fake source``` to ```WebRTC Development UI```. Can you help me to implement it?

[image]
[image]
[image]