mavlink/mavlink-camera-manager #273: Google Coral camera not detected on /dev/video3

steamfoundry, 2023-08-22:
The V4L compatible camera on /dev/video3 isn't detected currently.  

### Camera
 [Google Coral](https://coral.ai/products/camera/#tech-specs) connected via MIPI-CSI 

**Example Gstreamer command that will stream to QGroundControl (As UDP)**
sudo gst-launch-1.0 v4l2src device=/dev/video3 ! video/x-raw,width=640,height=480,framerate=30/1 ! vpuenc_h264 bitrate=500 ! rtph264pay ! udpsink host=192.168.1.103 port=5600 sync=false

### Board details

-     Board - NXP NavQPlus
-     Processor -NXP i.MX 8M Plus (4x Arm Cortex A53 cores)
-     OS: Ubuntu 22.04 on Yocto Linux 5.15 kernel - [Build 22.04.3 from Rudis Labs](https://github.com/rudislabs/navqplus-create3-images/releases/tag/v22.04.3)

Output from mavlink-camera-manager
```

2023-08-22T15:31:00.768533Z  INFO main ThreadId(01) src/logger/manager.rs:82: mavlink-camera-manager, version: 0.2.4-5e2a1f8, build date: 2023-08-21
2023-08-22T15:31:00.769195Z  INFO main ThreadId(01) src/logger/manager.rs:89: Starting at 2023-08-22T15:31:00
2023-08-22T15:31:00.769568Z DEBUG main ThreadId(01) src/logger/manager.rs:93: Command line call: ./mavlink-camera-manager -v --mavlink=udpout:10.0.0.2:14551 --enable-tracing-level-log-file
2023-08-22T15:31:00.769768Z DEBUG main ThreadId(01) src/logger/manager.rs:94: Command line input struct call: {
    "log-path": MatchedArg {
        occurs: 0,
        indices: [
            6,
        ],
        vals: [
            "./logs",
        ],
    },
    "enable-tracing-level-log-file": MatchedArg {
        occurs: 1,
        indices: [
            4,
        ],
        vals: [],
    },
    "rest-server": MatchedArg {
        occurs: 0,
        indices: [
            5,
        ],
        vals: [
            "0.0.0.0:6020",
        ],
    },
    "verbose": MatchedArg {
        occurs: 1,
        indices: [
            1,
        ],
        vals: [],
    },
    "mavlink": MatchedArg {
        occurs: 1,
        indices: [
            3,
        ],
        vals: [
            "udpout:10.0.0.2:14551",
        ],
    },
}
2023-08-22T15:31:00.770383Z DEBUG main ThreadId(01) src/settings/manager.rs:80: Using settings file: "/home/user/.config/mavlink-camera-manager/settings.json"
2023-08-22T15:31:00.771705Z DEBUG main ThreadId(01) init: src/stream/manager.rs:67: Starting video stream service.
2023-08-22T15:31:00.817495Z DEBUG main ThreadId(01) src/settings/manager.rs:148: Settings saved: ManagerStruct {
    file_name: "/home/user/.config/mavlink-camera-manager/settings.json",
    config: SettingsStruct {
        header: HeaderSettingsFile {
            name: "Camera Manager",
            version: 0,
        },
        mavlink_endpoint: Some(
            "udpout:10.0.0.2:14551",
        ),
        streams: [],
    },
}
2023-08-22T15:31:00.818828Z DEBUG SignallingServer ThreadId(02) run_main_loop: src/stream/webrtc/signalling_server.rs:70: Starting Signalling server on Url { scheme: "ws", cannot_be_a_base: false, username: "", password: None, host: Some(Ipv4(0.0.0.0)), port: Some(6021), path: "/", query: None, fragment: None }...
2023-08-22T15:31:00.826309Z DEBUG  SignallingServer ThreadId(02) run_main_loop:runner{endpoint=Url { scheme: "ws", cannot_be_a_base: false, username: "", password: None, host: Some(Ipv4(0.0.0.0)), port: Some(6021), path: "/", query: None, fragment: None }}: src/stream/webrtc/signalling_server.rs:91: Signalling server: listening on: 0.0.0.0:6021
2023-08-22T15:31:00.829553Z  WARN              main ThreadId(01) start_default: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:cap_devic\""
2023-08-22T15:31:00.848742Z  WARN              main ThreadId(01) start_default: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:m2m_devic\""
2023-08-22T15:31:00.854622Z ERROR              main ThreadId(01) start_default: src/video/video_source_local.rs:623: Faield to get device "/dev/video1": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-08-22T15:31:00.855150Z ERROR              main ThreadId(01) start_default: src/video/video_source_local.rs:623: Faield to get device "/dev/video0": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-08-22T15:31:00.855913Z DEBUG              main ThreadId(01) start_default: src/stream/manager.rs:139: Streams: []
2023-08-22T15:31:00.857553Z  INFO              main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/builder.rs:200: starting 4 workers
2023-08-22T15:31:00.857808Z  INFO              main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/server.rs:196: Actix runtime found; starting in Actix runtime
```
[mavlink-camera-manager.2023-08-22-15.log](https://github.com/mavlink/mavlink-camera-manager/files/12410469/mavlink-camera-manager.2023-08-22-15.log)

joaoantoniocardoso, 2023-08-23:
Hi @steamfoundry, the logs are not even mentioning the `/dev/video3` device, this is curious. 

Could you also provide:

- output of `ls -lah /dev/video*`
- output of `v4l2-ctl --all`

Thanks

steamfoundry, 2023-08-24:
```
ser@NAVQP:~$ ls -lah /dev/video*
crw-rw----+ 1 root video 81, 0 Aug 23 22:02 /dev/video0
crw-rw----+ 1 root video 81, 1 Aug 23 22:02 /dev/video1
crw-rw----+ 1 root video 81, 2 Aug 23 22:02 /dev/video2
crw-rw----+ 1 root video 81, 3 Aug 23 22:02 /dev/video3
user@NAVQP:~$ v4l2-ctl --all
Failed to open /dev/video0: Device or resource busy
```
Here is some more that might be helpful

Output of v4l2-ctl --list-devices
```
user@NAVQP:~$ sudo v4l2-ctl --list-devices --all
 ():
	/dev/v4l-subdev0

 ():
	/dev/v4l-subdev1

mxc-isi-cap (platform:32e00000.isi:cap_devic):
	/dev/video3

mxc-isi-m2m (platform:32e00000.isi:m2m_devic):
	/dev/video2

FSL Capture Media Device (platform:mxc-md):
	/dev/media0

Failed to open /dev/video0: Device or resource busy

```
Output to show the v4l subdevices
```
user@NAVQP:~$ ls -lah /dev/v4*
crw-rw----+ 1 root video 81, 5 Aug 23 22:02 /dev/v4l-subdev0
crw-rw----+ 1 root video 81, 6 Aug 23 22:02 /dev/v4l-subdev1
crw-rw----+ 1 root video 81, 7 Aug 23 22:02 /dev/v4l-subdev2

/dev/v4l:
total 0
drwxr-xr-x  3 root root   60 Aug 23 22:02 .
drwxr-xr-x 16 root root 7.6K Aug 24 05:52 ..
drwxr-xr-x  2 root root  120 Aug 23 22:02 by-path
```
ls of dev/v4l/by-path/
```
ser@NAVQP:~$ ls -lah /dev/v4l/by-path/
total 0
drwxr-xr-x 2 root root 120 Aug 23 22:02 .
drwxr-xr-x 3 root root  60 Aug 23 22:02 ..
lrwxrwxrwx 1 root root  12 Aug 23 22:02 platform-32c00000.bus:camera-video-index0 -> ../../video3
lrwxrwxrwx 1 root root  12 Aug 23 22:02 platform-32e00000.isi:m2m_device-video-index0 -> ../../video2
lrwxrwxrwx 1 root root  12 Aug 23 22:02 platform-vpu_v4l2-video-index0 -> ../../video0
lrwxrwxrwx 1 root root  12 Aug 23 22:02 platform-vpu_v4l2-video-index1 -> ../../video1
```

Yet more information
```
ser@NAVQP:~$ v4l2-ctl -d /dev/video0 -D 
Failed to open /dev/video0: Device or resource busy
user@NAVQP:~$ v4l2-ctl -d /dev/video1 -D 
Failed to open /dev/video1: Device or resource busy

user@NAVQP:~$ v4l2-ctl -d /dev/video2 -D 
Driver Info:
	Driver name      : mxc-isi-m2m
	Card type        : mxc-isi-m2m
	Bus info         : platform:32e00000.isi:m2m_devic
	Driver version   : 5.15.32
	Capabilities     : 0x84204000
		Video Memory-to-Memory Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04204000
		Video Memory-to-Memory Multiplanar
		Streaming
		Extended Pix Format
user@NAVQP:~$ v4l2-ctl -d /dev/video3 -D 
Driver Info:
	Driver name      : mxc-isi-cap
	Card type        : mxc-isi-cap
	Bus info         : platform:32e00000.isi:cap_devic
	Driver version   : 5.15.32
	Capabilities     : 0x84201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : mxc-md
	Model            : FSL Capture Media Device
	Serial           : 
	Bus info         : 
	Media version    : 5.15.32
	Hardware revision: 0x00000000 (0)
	Driver version   : 5.15.32
Interface Info:
	ID               : 0x03000014
	Type             : V4L Video
Entity Info:
	ID               : 0x00000012 (18)
	Name             : mxc_isi.0.capture
	Function         : V4L2 I/O
	Pad 0x01000013   : 0: Sink
	  Link 0x0200003f: from remote pad 0x100000e of entity 'mxc_isi.0' (Video Pixel Formatter): Data, Enabled
```

steamfoundry, 2023-08-24:
After a little more digging, I suspect these extra devices are extra functionality of the i.MX8MP processor.   
[some discussion here]https://community.nxp.com/t5/i-MX-Processors/iMX8MPlus-How-to-mix-two-video-sources-into-one-with-Gstreamer/m-p/1534910]

Because the image I'm using is Ubuntu on a Yocto kernel instead of a pure Yocto build, it's missing the drivers needed to fully identify the extra devices.

Here is a little more detail on them.
/dev/video0
```
root@NAVQP:/home/user# cat /sys/class/video4linux/video0/name

root@NAVQP:/home/user# cat /sys/class/video4linux/video0/uevent
MAJOR=81
MINOR=0
DEVNAME=video0
root@NAVQP:/home/user# cat /sys/class/video4linux/video0/device/uevent 
DRIVER=vsiv4l2
OF_NAME=vpu_v4l2
OF_FULLNAME=/vpu_v4l2
OF_COMPATIBLE_0=nxp,imx8m-vsiv4l2
OF_COMPATIBLE_N=1
MODALIAS=of:Nvpu_v4l2T(null)Cnxp,imx8m-vsiv4l2

```
/dev/video1
```
root@NAVQP:/home/user# cat /sys/class/video4linux/video1/name

root@NAVQP:/home/user# cat /sys/class/video4linux/video1/uevent
MAJOR=81
MINOR=1
DEVNAME=video1
root@NAVQP:/home/user# cat /sys/class/video4linux/video1/device/uevent 
DRIVER=vsiv4l2
OF_NAME=vpu_v4l2
OF_FULLNAME=/vpu_v4l2
OF_COMPATIBLE_0=nxp,imx8m-vsiv4l2
OF_COMPATIBLE_N=1
MODALIAS=of:Nvpu_v4l2T(null)Cnxp,imx8m-vsiv4l2

```
/dev/video2
```
root@NAVQP:/home/user# cat /sys/class/video4linux/video2/name
mxc_isi.0.m2m
root@NAVQP:/home/user# cat /sys/class/video4linux/video2/uevent
MAJOR=81
MINOR=2
DEVNAME=video2
root@NAVQP:/home/user# cat /sys/class/video4linux/video2/device/uevent 
DRIVER=isi-m2m
OF_NAME=m2m_device
OF_FULLNAME=/soc@0/bus@32c00000/camera/isi@32e00000/m2m_device
OF_COMPATIBLE_0=imx-isi-m2m
OF_COMPATIBLE_N=1
MODALIAS=of:Nm2m_deviceT(null)Cimx-isi-m2m
```
/dev/video3
```
root@NAVQP:/home/user# cat /sys/class/video4linux/video3/name
mxc_isi.0.capture
root@NAVQP:/home/user# cat /sys/class/video4linux/video3/uevent
MAJOR=81
MINOR=3
DEVNAME=video3
root@NAVQP:/home/user# cat /sys/class/video4linux/video3/device/uevent
DRIVER=mxc-md
OF_NAME=camera
OF_FULLNAME=/soc@0/bus@32c00000/camera
OF_COMPATIBLE_0=fsl,mxc-md
OF_COMPATIBLE_1=simple-bus
OF_COMPATIBLE_N=2
MODALIAS=of:NcameraT(null)Cfsl,mxc-mdCsimple-bus
```

joaoantoniocardoso, 2023-08-24:
Sorry for the delay responding you, that's great info, it indeed makes sense, `platform:32e00000.isi:cap_devic` is `/dev/video3`, so the MCM says it couldn't recognize it, and it's a fault on our identification part. What it means, however, is that the device is not a known type, thus, it should listed as a `VideoSourceLocalType::Unknown("platform:32e00000.isi:cap_devic")`, but we don't have a debug message there to help us.

It seems there's something wrong in the device discovery, by design it should work with `VideoSourceLocalType::Unknown` source types.

You might need to add a couple of debug for us to understand it better, for instance, we can start wit the following one:

```diff
diff --git a/src/video/video_source_local.rs b/src/video/video_source_local.rs
index 1aea91b..15c5b25 100644
--- a/src/video/video_source_local.rs
+++ b/src/video/video_source_local.rs
@@ -650,6 +650,7 @@ impl VideoSourceAvailable for VideoSourceLocal {
             cameras.push(VideoSourceType::Local(source));
         }
 
+        debug!("cameras available: {cameras:#?}");
         cameras
     }
 }
```

Thanks

steamfoundry, 2023-08-24:
Here was the output
```
2023-08-24T16:03:09.052782Z  INFO main ThreadId(01) src/logger/manager.rs:82: mavlink-camera-manager, version: 0.2.4-5e2a1f8, build date: 2023-08-24
2023-08-24T16:03:09.057842Z  INFO main ThreadId(01) src/logger/manager.rs:89: Starting at 2023-08-24T16:03:09
2023-08-24T16:03:09.058445Z DEBUG main ThreadId(01) src/logger/manager.rs:93: Command line call: /home/user/src/mavlink-camera-manager/target/debug/mavlink-camera-manager --mavlink=tcpout:127.0.0.1:5760 --verbose --reset --enable-tracing-level-log-file
2023-08-24T16:03:09.058770Z DEBUG main ThreadId(01) src/logger/manager.rs:94: Command line input struct call: {
    "rest-server": MatchedArg {
        occurs: 0,
        indices: [
            6,
        ],
        vals: [
            "0.0.0.0:6020",
        ],
    },
    "reset": MatchedArg {
        occurs: 1,
        indices: [
            4,
        ],
        vals: [],
    },
    "mavlink": MatchedArg {
        occurs: 1,
        indices: [
            2,
        ],
        vals: [
            "tcpout:127.0.0.1:5760",
        ],
    },
    "enable-tracing-level-log-file": MatchedArg {
        occurs: 1,
        indices: [
            5,
        ],
        vals: [],
    },
    "verbose": MatchedArg {
        occurs: 1,
        indices: [
            3,
        ],
        vals: [],
    },
    "log-path": MatchedArg {
        occurs: 0,
        indices: [
            7,
        ],
        vals: [
            "./logs",
        ],
    },
}
2023-08-24T16:03:09.062212Z DEBUG main ThreadId(01) src/settings/manager.rs:77: Settings reset, an empty settings will be loaded and stored as "/root/.config/mavlink-camera-manager/settings.json".
2023-08-24T16:03:09.062527Z  INFO main ThreadId(01) src/settings/manager.rs:107: The settings file "/root/.config/mavlink-camera-manager/settings.json" will be backed-up as "/root/.config/mavlink-camera-manager/settings.json.bak", and a new (empty) settings file will be created in its place.
2023-08-24T16:03:09.072351Z DEBUG main ThreadId(01) init: src/stream/manager.rs:67: Starting video stream service.
2023-08-24T16:03:09.297140Z DEBUG main ThreadId(01) src/settings/manager.rs:148: Settings saved: ManagerStruct {
    file_name: "/root/.config/mavlink-camera-manager/settings.json",
    config: SettingsStruct {
        header: HeaderSettingsFile {
            name: "Camera Manager",
            version: 0,
        },
        mavlink_endpoint: Some(
            "tcpout:127.0.0.1:5760",
        ),
        streams: [],
    },
}
2023-08-24T16:03:09.300365Z DEBUG SignallingServer ThreadId(02) run_main_loop: src/stream/webrtc/signalling_server.rs:70: Starting Signalling server on Url { scheme: "ws", cannot_be_a_base: false, username: "", password: None, host: Some(Ipv4(0.0.0.0)), port: Some(6021), path: "/", query: None, fragment: None }...
2023-08-24T16:03:09.309097Z DEBUG  SignallingServer ThreadId(02) run_main_loop:runner{endpoint=Url { scheme: "ws", cannot_be_a_base: false, username: "", password: None, host: Some(Ipv4(0.0.0.0)), port: Some(6021), path: "/", query: None, fragment: None }}: src/stream/webrtc/signalling_server.rs:91: Signalling server: listening on: 0.0.0.0:6021
2023-08-24T16:03:09.311140Z  WARN              main ThreadId(01) start_default: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:cap_devic\""
2023-08-24T16:03:09.330513Z  WARN              main ThreadId(01) start_default: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:m2m_devic\""
2023-08-24T16:03:09.336436Z ERROR              main ThreadId(01) start_default: src/video/video_source_local.rs:623: Faield to get device "/dev/video1": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-08-24T16:03:09.337213Z ERROR              main ThreadId(01) start_default: src/video/video_source_local.rs:623: Faield to get device "/dev/video0": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-08-24T16:03:09.337615Z DEBUG              main ThreadId(01) start_default: src/video/video_source_local.rs:653: cameras available: []
2023-08-24T16:03:09.338377Z DEBUG              main ThreadId(01) start_default: src/stream/manager.rs:139: Streams: []

```

joaoantoniocardoso, 2023-08-24:
Okay, the problem must be right in that function, let's try to see where:

```diff
diff --git a/src/video/video_source_local.rs b/src/video/video_source_local.rs
index 1aea91b..a124781 100644
--- a/src/video/video_source_local.rs
+++ b/src/video/video_source_local.rs
@@ -599,6 +599,7 @@ impl VideoSourceAvailable for VideoSourceLocal {
                     return cameras;
                 }
             };
+            debug!("read_dir: {read_dir:#?}");
 
             read_dir
                 .filter_map(|dir_entry| {
@@ -615,6 +616,7 @@ impl VideoSourceAvailable for VideoSourceLocal {
                 .filter(|entry_str| entry_str.starts_with("/dev/video"))
                 .collect::<Vec<String>>()
         };
+        debug!("cameras path: {cameras_path:#?}");
 
         for camera_path in &cameras_path {
             let device = match Device::with_path(camera_path) {
@@ -632,11 +634,13 @@ impl VideoSourceAvailable for VideoSourceLocal {
                     continue;
                 }
             };
+            debug!("caps: {caps:#?}");
 
             let typ = VideoSourceLocalType::from_str(&caps.bus);
 
             if let Err(error) = device.format() {
-                if error.kind() != std::io::ErrorKind::InvalidInput {
+                // if error.kind() != std::io::ErrorKind::InvalidInput
+                {
                     debug!("Failed to capture formats for device: {camera_path}\nError: {error:?}");
                 }
                 continue;
@@ -650,6 +654,7 @@ impl VideoSourceAvailable for VideoSourceLocal {
             cameras.push(VideoSourceType::Local(source));
         }
 
+        debug!("cameras available: {cameras:#?}");
         cameras
     }
 }
```

steamfoundry, 2023-08-24:
Build failed on the commented out if statement.   Was this correct?

```
 if let Err(error) = device.format() {
               // if error.kind() != std::io::ErrorKind::InvalidInput {
                debug!("Failed to capture formats for device: {camera_path}\nError: {error:?}");
//                }
                continue;
            }
```

The compile error was 
```
 if let Err(error) = device.format() {
               // if error.kind() != std::io::ErrorKind::InvalidInput {
                debug!("Failed to capture formats for device: {camera_path}\nError: {error:?}");
//                }
                continue;
            }
```

joaoantoniocardoso, 2023-08-24:
Sorry:

```diff
diff --git a/src/video/video_source_local.rs b/src/video/video_source_local.rs
index 1aea91b..78fe7a7 100644
--- a/src/video/video_source_local.rs
+++ b/src/video/video_source_local.rs
@@ -599,6 +599,7 @@ impl VideoSourceAvailable for VideoSourceLocal {
                     return cameras;
                 }
             };
+            debug!("read_dir: {read_dir:#?}");
 
             read_dir
                 .filter_map(|dir_entry| {
@@ -615,6 +616,7 @@ impl VideoSourceAvailable for VideoSourceLocal {
                 .filter(|entry_str| entry_str.starts_with("/dev/video"))
                 .collect::<Vec<String>>()
         };
+        debug!("cameras path: {cameras_path:#?}");
 
         for camera_path in &cameras_path {
             let device = match Device::with_path(camera_path) {
@@ -632,13 +634,14 @@ impl VideoSourceAvailable for VideoSourceLocal {
                     continue;
                 }
             };
+            debug!("caps: {caps:#?}");
 
             let typ = VideoSourceLocalType::from_str(&caps.bus);
 
             if let Err(error) = device.format() {
-                if error.kind() != std::io::ErrorKind::InvalidInput {
-                    debug!("Failed to capture formats for device: {camera_path}\nError: {error:?}");
-                }
+                // if error.kind() != std::io::ErrorKind::InvalidInput {
+                debug!("Failed to capture formats for device: {camera_path}\nError: {error:?}");
+                // }
                 continue;
             }
 
@@ -650,6 +653,7 @@ impl VideoSourceAvailable for VideoSourceLocal {
             cameras.push(VideoSourceType::Local(source));
         }
 
+        debug!("cameras available: {cameras:#?}");
         cameras
     }
 }
```

steamfoundry, 2023-08-24:
Here is the new log
```
023-08-24T17:33:44.614649Z  INFO main ThreadId(01) src/logger/manager.rs:82: mavlink-camera-manager, version: 0.2.4-5e2a1f8, build date: 2023-08-24
2023-08-24T17:33:44.621681Z  INFO main ThreadId(01) src/logger/manager.rs:89: Starting at 2023-08-24T17:33:44
2023-08-24T17:33:44.622274Z DEBUG main ThreadId(01) src/logger/manager.rs:93: Command line call: /home/user/src/mavlink-camera-manager/target/debug/mavlink-camera-manager --mavlink=tcpout:127.0.0.1:5760 --verbose --reset --enable-tracing-level-log-file
2023-08-24T17:33:44.622612Z DEBUG main ThreadId(01) src/logger/manager.rs:94: Command line input struct call: {
    "reset": MatchedArg {
        occurs: 1,
        indices: [
            4,
        ],
        vals: [],
    },
    "enable-tracing-level-log-file": MatchedArg {
        occurs: 1,
        indices: [
            5,
        ],
        vals: [],
    },
    "mavlink": MatchedArg {
        occurs: 1,
        indices: [
            2,
        ],
        vals: [
            "tcpout:127.0.0.1:5760",
        ],
    },
    "verbose": MatchedArg {
        occurs: 1,
        indices: [
            3,
        ],
        vals: [],
    },
    "rest-server": MatchedArg {
        occurs: 0,
        indices: [
            6,
        ],
        vals: [
            "0.0.0.0:6020",
        ],
    },
    "log-path": MatchedArg {
        occurs: 0,
        indices: [
            7,
        ],
        vals: [
            "./logs",
        ],
    },
}
2023-08-24T17:33:44.623448Z DEBUG main ThreadId(01) src/settings/manager.rs:77: Settings reset, an empty settings will be loaded and stored as "/root/.config/mavlink-camera-manager/settings.json".
2023-08-24T17:33:44.623668Z  INFO main ThreadId(01) src/settings/manager.rs:107: The settings file "/root/.config/mavlink-camera-manager/settings.json" will be backed-up as "/root/.config/mavlink-camera-manager/settings.json.bak", and a new (empty) settings file will be created in its place.
2023-08-24T17:33:44.626426Z DEBUG main ThreadId(01) init: src/stream/manager.rs:67: Starting video stream service.
2023-08-24T17:33:44.700183Z DEBUG main ThreadId(01) src/settings/manager.rs:148: Settings saved: ManagerStruct {
    file_name: "/root/.config/mavlink-camera-manager/settings.json",
    config: SettingsStruct {
        header: HeaderSettingsFile {
            name: "Camera Manager",
            version: 0,
        },
        mavlink_endpoint: Some(
            "tcpout:127.0.0.1:5760",
        ),
        streams: [],
    },
[... 22 lines cut ...]
2023-08-24T17:33:44.713239Z DEBUG  SignallingServer ThreadId(02) run_main_loop:runner{endpoint=Url { scheme: "ws", cannot_be_a_base: false, username: "", password: None, host: Some(Ipv4(0.0.0.0)), port: Some(6021), path: "/", query: None, fragment: None }}: src/stream/webrtc/signalling_server.rs:91: Signalling server: listening on: 0.0.0.0:6021
2023-08-24T17:33:44.718697Z  WARN              main ThreadId(01) start_default: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:cap_devic\""
2023-08-24T17:33:44.719100Z DEBUG              main ThreadId(01) start_default: src/video/video_source_local.rs:643: Failed to capture formats for device: /dev/video3
Error: Os { code: 22, kind: InvalidInput, message: "Invalid argument" }
2023-08-24T17:33:44.729759Z DEBUG              main ThreadId(01) start_default: src/video/video_source_local.rs:637: caps: Capabilities {
    driver: "mxc-isi-m2m",
    card: "mxc-isi-m2m",
    bus: "platform:32e00000.isi:m2m_devic",
    version: (
        5,
        15,
        32,
    ),
    capabilities: VIDEO_M2M_MPLANE | EXT_PIX_FORMAT | STREAMING,
}
2023-08-24T17:33:44.738742Z  WARN              main ThreadId(01) start_default: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:m2m_devic\""
2023-08-24T17:33:44.739199Z DEBUG              main ThreadId(01) start_default: src/video/video_source_local.rs:643: Failed to capture formats for device: /dev/video2
Error: Os { code: 22, kind: InvalidInput, message: "Invalid argument" }
2023-08-24T17:33:44.745044Z ERROR              main ThreadId(01) start_default: src/video/video_source_local.rs:625: Faield to get device "/dev/video1": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-08-24T17:33:44.745691Z ERROR              main ThreadId(01) start_default: src/video/video_source_local.rs:625: Faield to get device "/dev/video0": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-08-24T17:33:44.746065Z DEBUG              main ThreadId(01) start_default: src/video/video_source_local.rs:656: cameras available: []
2023-08-24T17:33:44.748173Z DEBUG              main ThreadId(01) start_default: src/stream/manager.rs:139: Streams: []
2023-08-24T17:33:44.750183Z  INFO              main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/builder.rs:200: starting 4 workers
2023-08-24T17:33:44.750494Z  INFO              main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/server.rs:196: Actix runtime found; starting in Actix runtime
```

joaoantoniocardoso, 2023-08-24:
Okay.
```
start_default: src/video/video_source_local.rs:643: Failed to capture formats for device: /dev/video3
Error: Os { code: 22, kind: InvalidInput, message: "Invalid argument" }
```

I need some time to investigate it, it is a direct response from the v4l library we are using.

joaoantoniocardoso, 2023-08-25:
Hi @steamfoundry, can you test [this PR](https://github.com/mavlink/mavlink-camera-manager/pull/277)? It's updating the v4l library, which might solve the problem.

steamfoundry, 2023-08-25:
Didn't seem to do it.
```
user@NAVQP:~/src/mavlink-camera-manager$ sudo ~/src/mavlink-camera-manager/target/debug/mavlink-camera-manager --mavlink=tcpout:127.0.0.1:5760  --verbose --reset --enable-tracing-level-log-file
[sudo] password for user: 
2023-08-25T22:53:54.929793Z  INFO main ThreadId(01) src/logger/manager.rs:82: mavlink-camera-manager, version: 0.2.4-3ec517a, build date: 2023-08-25
2023-08-25T22:53:54.948929Z  INFO main ThreadId(01) src/logger/manager.rs:89: Starting at 2023-08-25T22:53:54
2023-08-25T22:53:54.957270Z DEBUG main ThreadId(01) src/logger/manager.rs:93: Command line call: /home/user/src/mavlink-camera-manager/target/debug/mavlink-camera-manager --mavlink=tcpout:127.0.0.1:5760 --verbose --reset --enable-tracing-level-log-file
2023-08-25T22:53:54.960974Z DEBUG main ThreadId(01) src/logger/manager.rs:94: Command line input struct call: {
    "reset": MatchedArg {
        occurs: 1,
        indices: [
            4,
        ],
        vals: [],
    },
    "enable-tracing-level-log-file": MatchedArg {
        occurs: 1,
        indices: [
            5,
        ],
        vals: [],
    },
    "log-path": MatchedArg {
        occurs: 0,
        indices: [
            7,
        ],
        vals: [
            "./logs",
        ],
    },
    "mavlink": MatchedArg {
        occurs: 1,
        indices: [
            2,
        ],
        vals: [
            "tcpout:127.0.0.1:5760",
        ],
    },
    "verbose": MatchedArg {
        occurs: 1,
        indices: [
            3,
        ],
        vals: [],
    },
    "rest-server": MatchedArg {
        occurs: 0,
        indices: [
            6,
        ],
        vals: [
            "0.0.0.0:6020",
        ],
    },
}
2023-08-25T22:53:54.967766Z DEBUG main ThreadId(01) src/settings/manager.rs:77: Settings reset, an empty settings will be loaded and stored as "/root/.config/mavlink-camera-manager/settings.json".
2023-08-25T22:53:54.968060Z  INFO main ThreadId(01) src/settings/manager.rs:107: The settings file "/root/.config/mavlink-camera-manager/settings.json" will be backed-up as "/root/.config/mavlink-camera-manager/settings.json.bak", and a new (empty) settings file will be created in its place.
2023-08-25T22:53:54.987417Z DEBUG main ThreadId(01) init: src/stream/manager.rs:67: Starting video stream service.
2023-08-25T22:53:55.183652Z DEBUG main ThreadId(01) src/settings/manager.rs:148: Settings saved: ManagerStruct {
    file_name: "/root/.config/mavlink-camera-manager/settings.json",
    config: SettingsStruct {
        header: HeaderSettingsFile {
            name: "Camera Manager",
            version: 0,
        },
        mavlink_endpoint: Some(
            "tcpout:127.0.0.1:5760",
        ),
        streams: [],
    },
}
2023-08-25T22:53:55.192531Z DEBUG SignallingServer ThreadId(02) run_main_loop: src/stream/webrtc/signalling_server.rs:70: Starting Signalling server on Url { scheme: "ws", cannot_be_a_base: false, username: "", password: None, host: Some(Ipv4(0.0.0.0)), port: Some(6021), path: "/", query: None, fragment: None }...
2023-08-25T22:53:55.203126Z  WARN             main ThreadId(01) start_default: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:cap_devic\""
2023-08-25T22:53:55.208648Z DEBUG  SignallingServer ThreadId(02) run_main_loop:runner{endpoint=Url { scheme: "ws", cannot_be_a_base: false, username: "", password: None, host: Some(Ipv4(0.0.0.0)), port: Some(6021), path: "/", query: None, fragment: None }}: src/stream/webrtc/signalling_server.rs:91: Signalling server: listening on: 0.0.0.0:6021
2023-08-25T22:53:55.222575Z  WARN              main ThreadId(01) start_default: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:m2m_devic\""
2023-08-25T22:53:55.228556Z ERROR              main ThreadId(01) start_default: src/video/video_source_local.rs:671: Faield to get device "/dev/video1": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-08-25T22:53:55.229219Z ERROR              main ThreadId(01) start_default: src/video/video_source_local.rs:671: Faield to get device "/dev/video0": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-08-25T22:53:55.232323Z DEBUG              main ThreadId(01) start_default: src/stream/manager.rs:139: Streams: []
2023-08-25T22:53:55.247487Z  INFO              main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/builder.rs:200: starting 4 workers
2023-08-25T22:53:55.247863Z  INFO              main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/server.rs:196: Actix runtime found; starting in Actix runtime
^C2023-08-25T22:54:12.406814Z  INFO                        main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/server.rs:319: SIGINT received; starting forced shutdown
2023-08-25T22:54:12.407273Z  INFO actix-rt|system:0|arbiter:1 ThreadId(14) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/worker.rs:591: shutting down idle worker
2023-08-25T22:54:12.407331Z  INFO actix-rt|system:0|arbiter:3 ThreadId(16) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/worker.rs:591: shutting down idle worker
[... cut ...]
2023-08-25T22:54:12.407636Z DEBUG       actix-server acceptor ThreadId(17) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/accept.rs:309: paused accepting connections on 0.0.0.0:6020
2023-08-25T22:54:12.407908Z  INFO       actix-server acceptor ThreadId(17) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/accept.rs:143: accept thread stopped
```

[mavlink-camera-manager.2023-08-22-15(1).log](https://github.com/mavlink/mavlink-camera-manager/files/12443965/mavlink-camera-manager.2023-08-22-15.1.log)

joaoantoniocardoso, 2023-08-25:
Thanks, I'll work on a better solution using GStreamer instead of relying on that v4l library.

steamfoundry, 2023-08-28:
Not sure if this would be too small of a use case for you.  The NXP i.MX 8M Plus has a lot of support for hardware encoding/decoding that works with Gstreamer. 

[https://developer.ridgerun.com/wiki/index.php/IMX8/Multimedia/GStreamer_Support/Hardware-accelerated_plugins](https://community.nxp.com/pwmxy87654/attachments/pwmxy87654/imx-processors%40tkb/15/2/i.MX8GStreamerUserGuide.pdf)

NXP also wrote a user guide. [https://community.nxp.com/pwmxy87654/attachments/pwmxy87654/imx-processors%40tkb/15/2/i.MX8GStreamerUserGuide.pdf](https://community.nxp.com/pwmxy87654/attachments/pwmxy87654/imx-processors%40tkb/15/2/i.MX8GStreamerUserGuide.pdf)

If there was the choice using Gstreamer to enable hardware encoding that would be great.

joaoantoniocardoso, 2023-08-28:
Excellent, thanks for the docs, as we are already using GStreamer in our pipelines, I believe it's just a matter of using the correct plugins, like perhaps `v4l2h264enc` for the i.MX 8M.

We have a plan to allow users to use and configure and use transcoding in our pipelines, it's being discussed [here](https://github.com/mavlink/mavlink-camera-manager/issues/274), you are more than welcome to participate :)

joaoantoniocardoso, 2023-09-04:
Hi @steamfoundry, could you test [this PR](https://github.com/mavlink/mavlink-camera-manager/pull/281) and post the full log here? I still need to work on some details, but it should detect your camera and its formats using GStreamer's Device Monitor API.

Thanks

steamfoundry, 2023-09-04:
Verbose output
'''
user@NAVQP:~/src/mavlink-camera-manager$ sudo ~/src/mavlink-camera-manager/target/debug/mavlink-camera-manager --mavlink=tcpout:127.0.0.1:5760  --verbose --reset --enable-tracing-level-log-file
[sudo] password for user: 
2023-09-04T23:45:34.707000Z  INFO main ThreadId(01) src/logger/manager.rs:82: mavlink-camera-manager, version: 0.2.4-ab35a82, build date: 2023-09-04
2023-09-04T23:45:34.707774Z  INFO main ThreadId(01) src/logger/manager.rs:89: Starting at 2023-09-04T23:45:34
2023-09-04T23:45:34.708442Z DEBUG main ThreadId(01) src/logger/manager.rs:93: Command line call: /home/user/src/mavlink-camera-manager/target/debug/mavlink-camera-manager --mavlink=tcpout:127.0.0.1:5760 --verbose --reset --enable-tracing-level-log-file
2023-09-04T23:45:34.708754Z DEBUG main ThreadId(01) src/logger/manager.rs:94: Command line input struct call: {
    "rest-server": MatchedArg {
        occurs: 0,
        indices: [
            6,
        ],
        vals: [
            "0.0.0.0:6020",
        ],
    },
    "mavlink": MatchedArg {
        occurs: 1,
        indices: [
            2,
        ],
        vals: [
            "tcpout:127.0.0.1:5760",
        ],
    },
    "verbose": MatchedArg {
        occurs: 1,
        indices: [
            3,
        ],
        vals: [],
    },
    "reset": MatchedArg {
        occurs: 1,
        indices: [
            4,
        ],
        vals: [],
    },
    "enable-tracing-level-log-file": MatchedArg {
        occurs: 1,
        indices: [
            5,
        ],
        vals: [],
    },
    "log-path": MatchedArg {
        occurs: 0,
        indices: [
            7,
        ],
        vals: [
            "./logs",
        ],
    },
}

(mavlink-camera-manager:11757): GStreamer-CRITICAL **: 23:45:34.830: gst_element_message_full_with_details: assertion 'GST_IS_ELEMENT (element)' failed
2023-09-04T23:45:34.932419Z DEBUG main ThreadId(01) providers: src/video/gst_device_monitor.rs:64: Provider: "decklinkdeviceprovider"
2023-09-04T23:45:34.932783Z DEBUG main ThreadId(01) providers: src/video/gst_device_monitor.rs:64: Provider: "pulsedeviceprovider"
2023-09-04T23:45:34.933043Z DEBUG main ThreadId(01) providers: src/video/gst_device_monitor.rs:64: Provider: "uvch264deviceprovider"
2023-09-04T23:45:34.933424Z DEBUG main ThreadId(01) providers: src/video/gst_device_monitor.rs:64: Provider: "v4l2deviceprovider"
2023-09-04T23:45:34.933711Z DEBUG main ThreadId(01) providers: src/video/gst_device_monitor.rs:64: Provider: "alsadeviceprovider"
[src/main.rs:34] video::gst_device_monitor::providers().unwrap() = ()
[src/main.rs:37] &devices = [
    Device {
        inner: TypedObjectRef {
            inner: 0x0000aaaab7cef4c0,
            type: GstV4l2Device,
[... 769 lines cut ...]
2023-09-04T23:45:34.982566Z DEBUG main ThreadId(01) init: src/stream/manager.rs:67: Starting video stream service.
2023-09-04T23:45:34.983992Z DEBUG main ThreadId(01) src/settings/manager.rs:148: Settings saved: ManagerStruct {
    file_name: "/root/.config/mavlink-camera-manager/settings.json",
    config: SettingsStruct {
        header: HeaderSettingsFile {
            name: "Camera Manager",
            version: 0,
        },
        mavlink_endpoint: Some(
            "tcpout:127.0.0.1:5760",
        ),
        streams: [],
    },
}
2023-09-04T23:45:34.985449Z DEBUG SignallingServer ThreadId(02) run_main_loop: src/stream/webrtc/signalling_server.rs:70: Starting Signalling server on Url { scheme: "ws", cannot_be_a_base: false, username: "", password: None, host: Some(Ipv4(0.0.0.0)), port: Some(6021), path: "/", query: None, fragment: None }...
2023-09-04T23:45:34.991395Z DEBUG  SignallingServer ThreadId(02) run_main_loop:runner{endpoint=Url { scheme: "ws", cannot_be_a_base: false, username: "", password: None, host: Some(Ipv4(0.0.0.0)), port: Some(6021), path: "/", query: None, fragment: None }}: src/stream/webrtc/signalling_server.rs:91: Signalling server: listening on: 0.0.0.0:6021
2023-09-04T23:45:34.996852Z  WARN              main ThreadId(01) start_default: src/video/video_source_local.rs:73: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:cap_devic\""
2023-09-04T23:45:35.016165Z  WARN              main ThreadId(01) start_default: src/video/video_source_local.rs:73: msg="Unable to identify the local camera connection type, please report the problem: \"platform:32e00000.isi:m2m_devic\""
2023-09-04T23:45:35.022055Z ERROR              main ThreadId(01) start_default: src/video/video_source_local.rs:840: Faield to get device "/dev/video1": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-09-04T23:45:35.022697Z ERROR              main ThreadId(01) start_default: src/video/video_source_local.rs:840: Faield to get device "/dev/video0": Os { code: 16, kind: ResourceBusy, message: "Device or resource busy" }
2023-09-04T23:45:35.023474Z DEBUG              main ThreadId(01) start_default: src/stream/manager.rs:139: Streams: []
2023-09-04T23:45:35.025352Z  INFO              main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/builder.rs:200: starting 4 workers
2023-09-04T23:45:35.025697Z  INFO              main ThreadId(01) /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.2.0/src/server.rs:196: Actix runtime found; starting in Actix runtime

''

steamfoundry, 2023-09-04:
And the trace log 
[mavlink-camera-manager.2023-09-04-23.log](https://github.com/mavlink/mavlink-camera-manager/files/12517273/mavlink-camera-manager.2023-09-04-23.log)

joaoantoniocardoso, 2023-09-05:
@steamfoundry it seems all right, GStreamer seems to be able to detect your camera formats. Have you tried to open the page and create the stream? I'd like the log after you try, if possible :)

steamfoundry, 2023-09-07:
Sorry, I tried but it's still not showing the camera in my available screens.   I tried to add it manually but that failed too.  Let me gather the logs for you.

steamfoundry, 2023-09-07:
[mavlink-camera-manager.2023-09-07-20.log](https://github.com/mavlink/mavlink-camera-manager/files/12553231/mavlink-camera-manager.2023-09-07-20.log)

joaoantoniocardoso, 2024-04-27:
Hi @steamfoundry, have you done anything about this?
We have made many improvements since last year, are you interested in trying it again?
Thanks