mavlink/mavlink-camera-manager #351: Use intersink instead of proxysink and filesink

joaoantoniocardoso, 2024-03-08:
There might be an opportunity to get rid of some tee + queue, but the main idea is to try to separate all sinks from our sources using intersink instead of proxysink (used for UDP and thumbnail), and filesink (used for RTSP).

joaoantoniocardoso, 2024-11-11:
[image]

[source](https://gstconf.ubicast.tv/videos/splitting-gstreamer-pipelines)

abhishek202206, 2025-04-30:
Hi @joaoantoniocardoso I am struggling with issue related to filesink only. I am able to launch a UVC camera from MCM application but I need to start recording the video and capture still imagery as soon as command comes from GCS/QGC. The issue is for implementing filesink  I would need to use mp4mux which isn't supported on RUST. Is there a better way to initiate file recording or capturing image frames on MCM

joaoantoniocardoso, 2025-04-30:
Hi @abhishek202206, sorry for the misleading information in this issue. All the `filesink` mentioned here was a typo, and I've just replaced it with `shmsink`.

The `filesink` discussion should be kept [here](https://github.com/mavlink/mavlink-camera-manager/issues/436).