mavlink/mavlink-camera-manager #354: QGC doesn't display video stream

adolgarev, 2024-03-11:
Hello,

I followed instructions in the README and tried t3.12.7 and t3.12.6 releases but QGC doesn't automatically display the video stream. I create ball-Fake source h264 rtp stream and can add it successfully manually in QGC settings, QGC then displays the stream as expected. But automatically this doesn't happen. I see that mavlink-camera-manager sends heartbeats and QGC reacts to them and requests MAV_CMD_REQUEST_CAMERA_INFORMATION. In mavlink-camera-manager logs

```
2024-03-10T04:22:38.616028Z TRACE           MavReceiver ThreadId(13) receiver_loop: src/mavlink/manager.rs:104: Message received: MavHeader { system_id: 255, component_id: 190, sequence: 11 }, COMMAND_LONG(COMMAND_LONG_DATA { param1: 1.0, param2: 0.0, param3: 0.0, param4: 0.0, param5: 0.0, param6: 0.0, param7: 0.0, command: MAV_CMD_REQUEST_CAMERA_INFORMATION, target_system: 1, target_component: 100, confirmation: 0 })
```

But mavlink-camera-manager neither acknowledges the message not sends the information back. I can verify this by watching messages on mavproxy to which mavlink-camera-manager is connected. As a result QGC reports "Giving up requesting camera info from 1 100"

Do you know what can be the problem? Thanks.

See the full log [mavlink-camera-manager.2024-03-10-04.log](https://github.com/mavlink/mavlink-camera-manager/files/14552703/mavlink-camera-manager.2024-03-10-04.log) attached

joaoantoniocardoso, 2024-03-11:
Hi! Thanks for reporting.

> I create ball-Fake source h264 rtp stream and can add it successfully

You mean, _rtsp_? I just submitted a PR fixing it.

joaoantoniocardoso, 2024-03-11:
@adolgarev could you try the new version ([t3.12.8](https://github.com/mavlink/mavlink-camera-manager/releases/tag/t3.12.8))? Then we reopen if it is still not working.

Thanks!

joaoantoniocardoso, 2024-03-11:
@adolgarev I released a new version ([t3.12.9](https://github.com/mavlink/mavlink-camera-manager/releases/tag/t3.12.9)) because the previous fix wasn't enough for older GStreamer versions (like 1.23.4).

adolgarev, 2024-03-11:
@joaoantoniocardoso thank you for your response. Recent build produces the same result, please find details below.

I configure the stream like below

[image]

>>I create ball-Fake source h264 rtp stream and can add it successfully
> You mean, rtsp? I just submitted a PR fixing it.

I believe it is RTP over UDP

If I add stream manually

[image]

It displays ok

[image]

But if I let QGC discover stream by itself I see 

[image]

and no video 

[image]

In logs mavlink-camera-manager sends only heartbeats, no message ack and no response with camera caps.

[image]

[image]

System id and component id looks right. I wonder why mavlink-camera-manager doesn't respond.

The new log: 
[mavlink-camera-manager.2024-03-11-19.log](https://github.com/mavlink/mavlink-camera-manager/files/14564080/mavlink-camera-manager.2024-03-11-19.log)

joaoantoniocardoso, 2024-03-11:
Hi, I'm able to reproduce the issue, I'll investigate it.

joaoantoniocardoso, 2024-03-12:
@adolgarev can you test #357 ?

- [mavlink-camera-manager-aarch64 ](https://github.com/joaoantoniocardoso/mavlink-camera-manager/actions/runs/8240611457/artifacts/1316713104)

- [mavlink-camera-manager-armv7](https://github.com/joaoantoniocardoso/mavlink-camera-manager/actions/runs/8240611457/artifacts/1316713395) 

- [mavlink-camera-manager-linux-desktop](https://github.com/joaoantoniocardoso/mavlink-camera-manager/actions/runs/8240611457/artifacts/1316712645)

adolgarev, 2024-03-12:
It works! Thank yo very much.
[image]

There is one warning in QGC which is not fatal but maybe worth a look.
[image]

joaoantoniocardoso, 2024-03-13:
@adolgarev from my tests, some unresponsiveness on the mavlink camera manager side makes the messages from the QGC be interpreted and answered too late, thus failing to start the stream.

We are still working on that.

joaoantoniocardoso, 2024-03-13:
> @adolgarev from my tests, some unresponsiveness on the mavlink camera manager side makes the messages from the QGC be interpreted and answered too late, thus failing to start the stream.
>
> We are still working on that.

[this PR](https://github.com/mavlink/mavlink-camera-manager/pull/359) solves this.

The `Unable to load camera parameters from camera definition` is something that I saw during the fix, but I'm not seeing it anymore, so I believe it was related to the performance issue.

closed 2024-03-14 by patrickelectric

adolgarev, 2024-03-14:
Thanks! I'll give it a try

adolgarev, 2024-03-15:
Can you please make a build with the PR above? Thanks

joaoantoniocardoso, 2024-03-15:
@adolgarev Now the changes are in master:

[mavlink-camera-manager-aarch64](https://github.com/mavlink/mavlink-camera-manager/actions/runs/8294669784/artifacts/1329074570)
[mavlink-camera-manager-armv7](https://github.com/mavlink/mavlink-camera-manager/actions/runs/8294669784/artifacts/1329074622)
[mavlink-camera-manager-linux-desktop](https://github.com/mavlink/mavlink-camera-manager/actions/runs/8294669784/artifacts/1329065003)

adolgarev, 2024-03-18:
I still see `CameraControlLog: Unable to load camera parameters from camera definition` on QGC, I believe it is because I'm using fake source.

patrickelectric, 2024-03-18:
@adolgarev but the video is showing ? Or only the controls ?

DmitriyLugovoy, 2024-03-22:
hello!
I modified src/stream/pipeline/v4l_pipeline.rs::76
`VideoEncodeType::Yuyv => {
    format!(
        concat!(
            "v4l2src device={device} do-timestamp=true",
            " ! videoconvert",
            " ! x264enc",
            " ! h264parse",
            " ! capsfilter name={filter_name} caps=video/x-h264,stream-format=avc,alignment=au,width={width},height={height},framerate={interval_denominator}/{interval_numerator}",
            " ! tee name={video_tee_name} allow-not-linked=true",
            " ! rtph264pay aggregate-mode=zero-latency config-interval=10 pt=96",
            " ! tee name={rtp_tee_name} allow-not-linked=true"
        ),`
        and src/stream/sink/image_sink.rs::355
        `VideoEncodeType::Yuyv => {
    // For h264, we need to filter-out unwanted non-key frames here, before decoding it.
    let filter = gst::ElementFactory::make("identity")
        .property("drop-buffer-flags", gst::BufferFlags::DELTA_UNIT)
        .property("sync", false)
        .build()?;
    let decoder = gst::ElementFactory::make("avdec_h264")
        .property_from_str("lowres", "2") // (0) is 'full'; (1) is '1/2-size'; (2) is '1/4-size'
        .build()?;
    decoder.has_property("discard-corrupted-frames", None).then(|| decoder.set_property("discard-corrupted-frames", true));
    _transcoding_elements.push(filter);
    _transcoding_elements.push(decoder);
}
`
Now I'm able to stream from YUYV to QGC.

Also I updated src/mavlink/mavlink_camera.rs::279
    flags: mavlink::common::CameraCapFlags::CAMERA_CAP_FLAGS_HAS_VIDEO_STREAM |
                            mavlink::common::CameraCapFlags::CAMERA_CAP_FLAGS_CAN_CAPTURE_IMAGE_IN_VIDEO_MODE |
                            mavlink::common::CameraCapFlags::CAMERA_CAP_FLAGS_HAS_BASIC_ZOOM |
                            mavlink::common::CameraCapFlags::CAMERA_CAP_FLAGS_HAS_BASIC_FOCUS,
                            
what I need to do to see zoom and focus controls in QGC?
Will appreciate for your help!

DmitriyLugovoy, 2024-03-22:
here logs of QGC and camera_def.xml and camera-manager logs
[QGC_logs.txt](https://github.com/mavlink/mavlink-camera-manager/files/14720274/QGC_logs.txt)
[HP HD Camera: HP HD Camera_HP HD Camera: HP HD Camera_000.zip](https://github.com/mavlink/mavlink-camera-manager/files/14720295/HP.HD.Camera.HP.HD.Camera_HP.HD.Camera.HP.HD.Camera_000.zip)

joaoantoniocardoso, 2024-03-22:
> hello! I modified src/stream/pipeline/v4l_pipeline.rs::76 `VideoEncodeType::Yuyv => { format!( concat!( "v4l2src device={device} do-timestamp=true", " ! videoconvert", " ! x264enc", " ! h264parse", [...]

Cool, I'll check your logs soon, but can you make a draft PR with your changes?

The zoom should show up if it is shown on the front-end page hosted by MCM.. as in, it will be shown if the camera has the zoom as a UCV/V4L2 control. (it also is possible that we are doing something wrong and not detecting that control for that camera somehow..)

DmitriyLugovoy, 2024-03-22:
ERROR: Permission to mavlink/mavlink-camera-manager.git denied to DmitriyLugovoy.

joaoantoniocardoso, 2024-03-22:
> ERROR: Permission to mavlink/mavlink-camera-manager.git denied to DmitriyLugovoy.

You need to fork the project, sorry =)

adolgarev, 2024-03-22:
> @adolgarev but the video is showing ? Or only the controls ?

Yes, video is showing

joaoantoniocardoso, 2024-03-22:
> > @adolgarev but the video is showing ? Or only the controls ?
> 
> Yes, video is showing

@adolgarev good, so I won't bother looking into that. Feel free to create new issues if you find anything else, or if you want to discuss new features, thanks!

tranhien1612, 2024-09-19:
@adolgarev @joaoantoniocardoso  i will take you the steps i did to display the video on QGC. I don't know if it's true or not, so I would like to have steps to accomplish that task. I tried with some my step below but it`s not work.
1. sudo ./mavlink-camera-manager --mavlink=tcpout:0.0.0.0:14444 --verbose
[image]

2. I config "Camera: Fake Source" and it had some problems.
[image]

Please help me to be able to watch videos via QGC!
Thanks!

joaoantoniocardoso, 2024-09-19:
Hi @adolgarev, can you post the output of the following terminal command?

```bash
gst-launch-1.0 --version
```

We strongly advise to use the newest GStreamer (some after 1.24), but 1.22 should not give that error.

tranhien1612, 2024-09-20:
@joaoantoniocardoso 
I checked my gst version as below, maybe the error is caused by low version.
```
gst-launch-1.0 version 1.20.1
GStreamer 1.16.3
https://launchpad.net/distros/ubuntu/+source/gstreamer1.0
```

joaoantoniocardoso, 2024-09-20:
@adolgarev Probably yes. I don't plan to support old GStreamer, so let me know if you need help getting the current version for your platform.

Also, I should add a check in the code and add this information to the readme.

tranhien1612, 2024-09-23:
@joaoantoniocardoso I am trying to display image from ```WebCam``` to ```WebRTC Development UI```. I see it doesn't display the information, is this related to the old version of GStreamer?

[image]
[image]
[image]

tranhien1612, 2024-09-23:
@joaoantoniocardoso 
I have installed gst according to the link: [gsteamer-1.22.0](https://www.linuxfromscratch.org/blfs/view/11.3/multimedia/gstreamer10.html) , after installation the version of gst is 1.22 as shown below.

```
gst-launch-1.0 version 1.22.0
GStreamer 1.22.0
https://www.linuxfromscratch.org/blfs/view/11.3/
```
After installing gst I cleaned by ```cargo clean`` and rebuilt the source code by ```cargo build``` but an error occurred.

[image]
[image]

tranhien1612, 2024-09-23:
@joaoantoniocardoso 
I have installed gst according to the link: [gsteamer-1.22.0](https://www.linuxfromscratch.org/blfs/view/11.3/multimedia/gstreamer10.html) , after installation the version of gst is 1.22 as shown below.

```
gst-launch-1.0 version 1.22.9
GStreamer 1.22.9
Unknown package origin
```
then i rebuilt the source code and tried again:

[image]

joaoantoniocardoso, 2024-09-23:
> @joaoantoniocardoso I am trying to display image from `WebCam` to `WebRTC Development UI`. I see it doesn't display the camera and video information in web, is this related to the old version of GStreamer?
> 
> ![Screenshot from 2024-09-23 08-53-22](https://private-user-images.githubusercontent.com/69721832/369734496-e739b838-a0e4-45ee-8da6-1c7ba1a27866.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxMDAyMjIsIm5iZiI6MTcyNzA5OTkyMiwicGF0aCI6Ii82OTcyMTgzMi8zNjk3MzQ0OTYtZTczOWI4MzgtYTBlNC00NWVlLThkYTYtMWM3YmExYTI3ODY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTIzVDEzNTg0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk3MjZmMjc2NDlkY2U2OGRjYzVkYjI3Y2NkMTJlMjI3ODI2ZWIyYzc1MWFkY2VhODNiZTlkOTVhYTRmMWM2ODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.R9IF48wcaHcUizNAwuQ238Jojvs1B0FOeFuChzaubEQ) ![Screenshot from 2024-09-23 08-53-18](https://private-user-images.githubusercontent.com/69721832/369734491-90870b45-dd37-4ddb-973b-cbeae208dc1d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxMDAyMjIsIm5iZiI6MTcyNzA5OTkyMiwicGF0aCI6Ii82OTcyMTgzMi8zNjk3MzQ0OTEtOTA4NzBiNDUtZGQzNy00ZGRiLTk3M2ItY2JlYWUyMDhkYzFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTIzVDEzNTg0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJlNGNhMzM3NTdjODZkMDUxZTYxMDgwZGU4ZDI2ZWMwYmQ2MjZhMjNkYjdmZWUyMzY0N2UxYWQ2NzQ2ZGQyMjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.IiztQEQmmOF_D31xeClLNYQnkFWkoS2pgSpBrS0Px5c) ![Screenshot from 2024-09-23 08-53-25](https://private-user-images.githubusercontent.com/69721832/369734498-9ea299de-ac2e-4b10-a06e-e1a007269a75.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcxMDAyMjIsIm5iZiI6MTcyNzA5OTkyMiwicGF0aCI6Ii82OTcyMTgzMi8zNjk3MzQ0OTgtOWVhMjk5ZGUtYWMyZS00YjEwLWEwNmUtZTFhMDA3MjY5YTc1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTIzVDEzNTg0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI0MGM3ODUxNjQ4MWM0NWRlYWE5Yjc4ODdmNzgzNjBmNzg3OGNmZTQ0M2Q3MzRmYjVkNTQ0MDg1Nzc0NDViZjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.XH4FinF8C7BTEmqFh3x4Odk8xFdPv4kCSIoFeewBkIo)

Hi! The WebRTC only works with h264, not YUYV, and we don't support encoding yet.

> then i rebuilt the source code and tried again:

Good, that version should be enough, although 1.22 can have some thread leaks on the WebRTC, depending on the platform and OS, the safest is 1.24.

The error is because in mentioned guide, it is not building most of the plugins. Here is the config I use:

```bash
meson build\
    --buildtype=release\
    --strip\
    -D bad=enabled\
    -D devtools=enabled\
    -D doc=disabled\
    -D ges=disabled\
    -D gpl=enabled\
    -D gst-plugins-bad:openh264=disabled\
    -D gst-plugins-bad:rtp=enabled\
    -D gst-plugins-base:tcp=enabled\
    -D gst-plugins-good:cairo=disabled\
    -D gst-plugins-good:jpeg=enabled\
    -D gst-plugins-good:rtsp=enabled\
    -D gst-plugins-good:udp=enabled\
    -D gst-plugins-good:v4l2=enabled\
    -D gst-plugins-good:vpx=enabled\
    -D gst-plugins-ugly:x264=enabled\
    -D gst-rtsp-server:examples=enabled\
    -D introspection=disabled\
    -D libav=enabled\
    -D nls=disabled\
    -D orc=disabled\
    -D python=disabled\
    -D qt5=disabled\
    -D rs=disabled\
    -D rtsp_server=enabled\
    -D tests=disabled\
    -D tls=enabled\
    -D ugly=enabled
```

Thanks!

PR #355 (2024-03-11): src: stream: rtsp: shmsrc needs to do-timestamp
Fixes #354 

This is partially a revert of #332
PR #356 (2024-03-11): src: stream: rtsp: shmsrc needs to wait for connection
This is also needed to fix #354 depending on the GStreamer version.
PR #357 (2024-03-12): grc: mavlink: Fix mavlink camera compoenent never answering QGC
The problem is that the added delay was causing the subscription to be always late.

Closes #354
PR #359 (2024-03-14): Fix mavlink unresponsiveness / lateness.
Finally, the clogging is solved.

lesson learned: Do not sleep on network-receiving, or channel-receiving loops.

The CPU usage on RPI4 spikes +6% when exchanging information back and forth with the GCS.

- [x] Tested locally
- [x] Tested with the ROV (RPI 4)

Closes #354