mavlink/mavlink-camera-manager #363: RTSP streams failing for JPEG and YUYV

joaoantoniocardoso, 2024-03-15:
From our manual testing of the current master (2b1dc02dd4152af3597550f916b89cf24871e1cb):

> RTSP streams for JPEG/YUYV are behaving inconsistently both for v4l and testsrc, and sometimes die immediately or after a couple of seconds, entering some locked state

lingaspeakingspider, 2024-05-19:
Facing the same problem here. What video streaming method works for now?

joaoantoniocardoso, 2024-05-19:
> Facing the same problem here. What video streaming method works for now?

- RTSP, UDP (plus WebRTC) for H264
- UDP for JPEG
- UDP for YUYV, as long as the receiver uses the SDP file

lingaspeakingspider, 2024-05-20:
Can QGC display UDP+JPEG?

lingaspeakingspider, 2024-05-20:
Nope, QGC doesn't show any video output. Only the record button

joaoantoniocardoso, 2024-05-20:
UDP+JPEG seems to be only working when receiving with MPV or VLC when using the SDP file.

joaoantoniocardoso, 2024-07-19:
Fixed by #426 and #427.

closed 2024-07-19 by joaoantoniocardoso