mavlink/mavlink-camera-manager #366: mavlink-camera-manager: command not found

tranhien1612, 2024-03-19:
I followed this guidle : 
'''https://github.com/mavlink/mavlink-camera-manager?tab=readme-ov-file#how-to-build-it'''
After build successfully, i run the command '''mavlink-camera-manager --mavlink=tcpout:0.0.0.0:14000 --verbose''' but there is a error:
'''
ras@ras:~ $ mavlink-camera-manager --mavlink=tcpout:0.0.0.0:14000 --verbose
-bash: mavlink-camera-manager: command not found
'''

pls help me fix it

patrickelectric, 2024-03-19:
Remember that you can download the binaries as described in the README.
If you did build the project successfully, you should be able to run it with `cargo run` or `./target/release/mavlink-camera-manager` after running `cargo build --release`

tranhien1612, 2024-03-19:
@patrickelectric 
I run ```cargo build --release``` successfully but when i run ```cargo run ``` an error has occurred
```
   Compiling mavlink-camera-manager v0.2.4 (/home/raspi/mavlink-camera-manager)
    Finished dev [unoptimized + debuginfo] target(s) in 20m 45s
     Running `target/debug/mavlink-camera-manager`
2024-03-19T03:58:56.247869Z  INFO main ThreadId(01) src/logger/manager.rs:84: mavlink-camera-manager, version: 0.2.4-2b1dc02, build date: 2024-03-19
2024-03-19T03:58:56.248667Z  INFO main ThreadId(01) src/logger/manager.rs:91: Starting at 2024-03-19T03:58:56
2024-03-19T03:58:56.261062Z  WARN main ThreadId(01) src/settings/manager.rs:128: Failed to load settings file "/home/raspi/.config/mavlink-camera-manager/settings.json". Reason: No such file or directory (os error 2)
thread 'main' panicked at src/mavlink/manager.rs:37:51:
No configured mavlink endpoint
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

patrickelectric, 2024-03-20:
@joaoantoniocardoso any idea what is going on ?

joaoantoniocardoso, 2024-03-20:
Yes, this is a known bug... Sorry. Run once with the `--reset` argument, and it should be fine.

joaoantoniocardoso, 2024-04-11:
Closed by #370.

Let me know if there is anything else.

closed 2024-04-11 by joaoantoniocardoso