mavlink/mavlink-camera-manager #371: Management interface not working when MCM fails to recognize v4l2 video sources

justas-, 2024-03-28:
I am trying to install MCM on an embedded system. I can access the management interface, but it does not allow me to create any streams. The only visible UI elements are heading "Streams" and button "WebRTC website".  The MCM verbose log is attached:

root@ucm-imx8m-plus:~# mavlink-camera-manager-aarch64 --verbose
2022-04-28T18:45:42.466321Z  INFO main ThreadId(01) src/logger/manager.rs:84: mavlink-camera-manager, version: 0.2.4-2b1dc02, build date: 2024-03-16
2022-04-28T18:45:42.466544Z  INFO main ThreadId(01) src/logger/manager.rs:91: Starting at 2022-04-28T18:45:42
2022-04-28T18:45:42.466608Z DEBUG main ThreadId(01) src/logger/manager.rs:95: Command line call: mavlink-camera-manager-aarch64 --verbose
2022-04-28T18:45:42.466677Z DEBUG main ThreadId(01) src/logger/manager.rs:96: Command line input struct call: Args {
    mavlink: None,
    default_settings: None,
    reset: false,
    rest_server: "0.0.0.0:6020",
    stun_server: "stun://0.0.0.0:3478",
    signalling_server: "ws://0.0.0.0:6021",
    verbose: true,
    gst_feature_rank: [],
    log_path: Some(
        "./logs",
    ),
    enable_tracing_level_log_file: false,
    vehicle_ddns: None,
    enable_tracy: false,
    enable_thread_counter: false,
    enable_webrtc_task_test: None,
    mavlink_system_id: 1,
}
2022-04-28T18:45:42.466808Z DEBUG main ThreadId(01) src/settings/manager.rs:80: Using settings file: "/home/root/.config/mavlink-camera-manager/settings.json"
2022-04-28T18:45:42.467903Z DEBUG main ThreadId(01) init: src/stream/manager.rs:68: Starting video stream service.
2022-04-28T18:45:42.519060Z DEBUG main ThreadId(01) default: src/stream/webrtc/signalling_server.rs:70: Starting SignallingServer task...
2022-04-28T18:45:42.519606Z DEBUG tokio-runtime-worker ThreadId(05) src/stream/webrtc/signalling_server.rs:73: SignallingServer task started!
2022-04-28T18:45:42.519855Z DEBUG tokio-runtime-worker ThreadId(05) runner{endpoint=Url { scheme: "ws", cannot_be_a_base: false, username: "", password: None, host: Some(Ipv4(0.0.0.0)), port: Some(6021), path: "/", query: None, fragment: None }}: src/stream/webrtc/signalling_server.rs:98: Signalling server: listening on: 0.0.0.0:6021
2022-04-28T18:45:42.524741Z  WARN                 main ThreadId(01) start_default: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:vsi_v4l2dec\""
2022-04-28T18:45:42.530433Z  WARN                 main ThreadId(01) start_default: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:vsi_v4l2enc\""
2022-04-28T18:45:42.530874Z DEBUG                 main ThreadId(01) start_default: src/stream/manager.rs:140: Streams: []
2022-04-28T18:45:42.531679Z  INFO                 main ThreadId(01) /cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.3.0/src/builder.rs:240: starting 4 workers
2022-04-28T18:45:42.531780Z  INFO                 main ThreadId(01) /cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-server-2.3.0/src/server.rs:197: Tokio runtime found; starting in existing Tokio runtime
2022-04-28T18:45:43.469056Z DEBUG          MavReceiver ThreadId(13) receiver_loop:connect{address="tcpout:127.0.0.1:5760"}: src/mavlink/manager.rs:234: Connecting...
2022-04-28T18:45:43.469474Z  INFO          MavReceiver ThreadId(13) receiver_loop:connect{address="tcpout:127.0.0.1:5760"}: src/mavlink/manager.rs:238: Successfully connected
2022-04-28T18:45:48.691912Z  INFO actix-server worker 0 ThreadId(14) /cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-web-4.5.1/src/middleware/logger.rs:420: 192.168.137.1 "GET / HTTP/1.1" 200 21057 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0" 0.000381
2022-04-28T18:45:48.771815Z  INFO actix-server worker 0 ThreadId(14) /cargo/registry/src/index.crates.io-6f17d22bba15001f/actix-web-4.5.1/src/middleware/logger.rs:420: 192.168.137.1 "GET /vue.js HTTP/1.1" 200 542311 "http://192.168.137.50:6020/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0" 0.023959
2022-04-28T18:45:48.848328Z  WARN actix-server worker 0 ThreadId(14) HTTP request{http.method=GET http.route=/v4l http.flavor=1.1 http.scheme=http http.host=192.168.137.50:6020 http.client_ip=192.168.137.1 http.user_agent=Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0 http.target=/v4l otel.name=HTTP GET /v4l otel.kind="server" request_id=ebcaa6b2-0dd6-4ee1-b2e8-038eb58778e9}: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:vsi_v4l2dec\""
2022-04-28T18:45:48.855200Z  WARN actix-server worker 0 ThreadId(14) HTTP request{http.method=GET http.route=/v4l http.flavor=1.1 http.scheme=http http.host=192.168.137.50:6020 http.client_ip=192.168.137.1 http.user_agent=Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0 http.target=/v4l otel.name=HTTP GET /v4l otel.kind="server" request_id=ebcaa6b2-0dd6-4ee1-b2e8-038eb58778e9}: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:vsi_v4l2enc\""
2022-04-28T18:45:48.860008Z  WARN actix-server worker 0 ThreadId(14) HTTP request{http.method=GET http.route=/v4l http.flavor=1.1 http.scheme=http http.host=192.168.137.50:6020 http.client_ip=192.168.137.1 http.user_agent=Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0 http.target=/v4l otel.name=HTTP GET /v4l otel.kind="server" request_id=ebcaa6b2-0dd6-4ee1-b2e8-038eb58778e9}: src/video/video_source_local.rs:295: Failed to get framesizes from format Description { index: 0, typ: 1, flags: (empty), description: "Y/CbCr 4:2:0", fourcc: FourCC { repr: [78, 86, 49, 50] } } for device "/dev/video1": Os {
    code: 25,
    kind: Uncategorized,
    message: "Inappropriate ioctl for device",
}
2022-04-28T18:45:48.860230Z  WARN actix-server worker 0 ThreadId(14) HTTP request{http.method=GET http.route=/v4l http.flavor=1.1 http.scheme=http http.host=192.168.137.50:6020 http.client_ip=192.168.137.1 http.user_agent=Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0 http.target=/v4l otel.name=HTTP GET /v4l otel.kind="server" request_id=ebcaa6b2-0dd6-4ee1-b2e8-038eb58778e9}: src/video/video_source_local.rs:295: Failed to get framesizes from format Description { index: 1, typ: 1, flags: (empty), description: "VSI DTRC", fourcc: FourCC { repr: [68, 84, 82, 67] } } for device "/dev/video1": Os {
    code: 25,
    kind: Uncategorized,
    message: "Inappropriate ioctl for device",
}
2022-04-28T18:45:48.860336Z  WARN actix-server worker 0 ThreadId(14) HTTP request{http.method=GET http.route=/v4l http.flavor=1.1 http.scheme=http http.host=192.168.137.50:6020 http.client_ip=192.168.137.1 http.user_agent=Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0 http.target=/v4l otel.name=HTTP GET /v4l otel.kind="server" request_id=ebcaa6b2-0dd6-4ee1-b2e8-038eb58778e9}: src/video/video_source_local.rs:295: Failed to get framesizes from format Description { index: 2, typ: 1, flags: (empty), description: "P010", fourcc: FourCC { repr: [80, 48, 49, 48] } } for device "/dev/video1": Os {
    code: 25,
    kind: Uncategorized,
    message: "Inappropriate ioctl for device",
}
2022-04-28T18:45:49.206911Z  WARN actix-server worker 0 ThreadId(14) HTTP request{http.method=GET http.route=/v4l http.flavor=1.1 http.scheme=http http.host=192.168.137.50:6020 http.client_ip=192.168.137.1 http.user_agent=Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0 http.target=/v4l otel.name=HTTP GET /v4l otel.kind="server" request_id=ebcaa6b2-0dd6-4ee1-b2e8-038eb58778e9}: src/video/video_source_local.rs:295: Failed to get framesizes from format Description { index: 3, typ: 1, flags: (empty), description: "NV12 10Bit", fourcc: FourCC { repr: [78, 86, 88, 50] } } for device "/dev/video1": Os {
    code: 25,
    kind: Uncategorized,
[... cut ...]  (KHTML, like Gecko) Chrome/122.0.0.0 Safari/537.36 Edg/122.0.0.0 http.target=/v4l otel.name=HTTP GET /v4l otel.kind="server" request_id=4dd3e6d3-08a3-40ec-bd9c-9faf3573a05a}: src/video/video_source_local.rs:72: msg="Unable to identify the local camera connection type, please report the problem: \"platform:vsi_v4l2enc\""
thread 'actix-server worker 1' panicked at /cargo/registry/src/index.crates.io-6f17d22bba15001f/v4l-0.14.0/src/control.rs:172:45:
called `Result::unwrap()` on an `Err` value: ()

joaoantoniocardoso, 2024-03-28:
Hi! Thanks for reporting. This seems to be an error originating from inside our v4l dependency: `src/control.rs:172:45`.

Could you please run it pretending `RUST_BACKTRACE=FULL` so we can better locate the error?

justas-, 2024-03-28:
Log file with RUST_BACKTRACE=FULL attached. Is there a way for me to ignore v4l2 devices, so I can test RTSP integration?

[mcm-log-ucm.txt](https://github.com/mavlink/mavlink-camera-manager/files/14794684/mcm-log-ucm.txt)

joaoantoniocardoso, 2024-04-05:
Hi @justas-,

The panic is being caused by [this unwrap in control.rs:172:45](https://github.com/raymanfx/libv4l-rs/blob/v4l-0.14.0/src/control.rs#L172).

Could you give us the output of the following command for each `DEVICE` in `/dev/video*` (probably `/dev/video0` and `/dev/video1`)?

```bash
v4l2-ctrl --list-ctrls-menus --device=$DEVICE
```

With that I believe we can quickly fix it.

justas-, 2024-04-05:
@joaoantoniocardoso thanks for looking into this. As requested:

Video0:
```
root@ucm-imx8m-plus:~# v4l2-ctl --list-ctrls-menus --device=/dev/video0

User Controls

                         rotate 0x00980922 (int)    : min=0 max=270 step=90 default=0 value=0 flags=modify-layout
  min_number_of_capture_buffers 0x00980927 (int)    : min=1 max=16 step=1 default=1 value=1 flags=read-only, volatile
   min_number_of_output_buffers 0x00980928 (int)    : min=1 max=16 step=1 default=1 value=1 flags=read-only, volatile
      get_max_roi_region_number 0x009819b2 (int)    : min=0 max=8 step=1 default=0 value=8 flags=read-only, volatile
   vsi_priv_v4l2_roi_params_set 0x009819b3 (unknown): type=164 value=unsupported payload type flags=has-payload
     get_max_ipcm_region_number 0x009819b4 (int)    : min=0 max=2 step=1 default=0 value=2 flags=read-only, volatile
  vsi_priv_v4l2_ipcm_params_set 0x009819b5 (unknown): type=165 value=unsupported payload type flags=has-payload

Codec Controls

                 video_b_frames 0x009909ca (int)    : min=0 max=0 step=1 default=0 value=0 flags=update
                 video_gop_size 0x009909cb (int)    : min=1 max=2147483647 step=1 default=30 value=30
             video_bitrate_mode 0x009909ce (menu)   : min=0 max=1 default=0 value=0 (Variable Bitrate) flags=update
                                0: Variable Bitrate
                                1: Constant Bitrate
                  video_bitrate 0x009909cf (int)    : min=10000 max=288000000 step=1 default=2097152 value=2097152
frame_level_rate_control_enable 0x009909d7 (bool)   : default=0 value=0
           sequence_header_mode 0x009909d8 (menu)   : min=0 max=1 default=1 value=1 (Joined With 1st Frame)
                                0: Separate Buffer
                                1: Joined With 1st Frame
     h264_mb_level_rate_control 0x009909da (bool)   : default=0 value=0
       number_of_mbs_in_a_slice 0x009909dc (int)    : min=1 max=120 step=1 default=1 value=1
      slice_partitioning_method 0x009909dd (menu)   : min=0 max=1 default=0 value=0 (Single)
                                0: Single
                                1: Max Macroblocks
         repeat_sequence_header 0x009909e2 (bool)   : default=1 value=1
                force_key_frame 0x009909e5 (button) : value=0 flags=write-only, execute-on-write
          h264_i_frame_qp_value 0x00990a5e (int)    : min=-1 max=51 step=1 default=30 value=30
          h264_p_frame_qp_value 0x00990a5f (int)    : min=-1 max=51 step=1 default=30 value=30
          h264_b_frame_qp_value 0x00990a60 (int)    : min=-1 max=51 step=1 default=30 value=30
          h264_minimum_qp_value 0x00990a61 (int)    : min=0 max=51 step=1 default=0 value=0
          h264_maximum_qp_value 0x00990a62 (int)    : min=0 max=51 step=1 default=51 value=51
           h264_cpb_buffer_size 0x00990a64 (int)    : min=0 max=288000000 step=1 default=0 value=0
                     h264_level 0x00990a67 (menu)   : min=0 max=15 default=14 value=14 (5)
                                0: 1
                                1: 1b
                                2: 1.1
                                3: 1.2
                                4: 1.3
                                5: 2
                                6: 2.1
                                7: 2.2
                                8: 3
                                9: 3.1
                                10: 3.2
                                11: 4
                                12: 4.1
                                13: 4.2
                                14: 5
                                15: 5.1
                   h264_profile 0x00990a6b (menu)   : min=0 max=16 default=0 value=0 (Baseline)
                                0: Baseline
                                1: Constrained Baseline
                                2: Main
                                3: Extended
                                4: High
                                5: High 10
                                6: High 422
                                7: High 444 Predictive
                                8: High 10 Intra
                                9: High 422 Intra
                                10: High 444 Intra
[... 63 lines cut ...]
                                8: High 10 Intra
                                9: High 422 Intra
                                10: High 444 Intra
                                11: CAVLC 444 Intra
                                12: Scalable Baseline
                                13: Scalable High
                                14: Scalable High Intra
                                15: Stereo High
                                16: Multiview High
                    vp8_profile 0x00990aff (menu)   : min=0 max=3 default=0 value=0 (0)
                                0: 0
                                1: 1
                                2: 2
                                3: 3
                    vp9_profile 0x00990b00 (menu)   : min=0 max=3 default=0 value=0 (0)
                                0: 0
                                1: 1
                                2: 2
                                3: 3
                   hevc_profile 0x00990b67 (menu)   : min=0 max=2 default=0 value=0 (Main)
                                0: Main
                                1: Main Still Picture
                                2: Main 10

```

joaoantoniocardoso, 2024-04-08:
Just so we can both track it, the issue was created [here](https://github.com/raymanfx/libv4l-rs/issues/98). Feel free to interact there, they might need more information to help us.

PR #379 (2024-04-14): Fix v4l panics
Closes #371
closed 2024-04-14 by patrickelectric