mavlink/mavlink-camera-manager #372: Add support to rpicamsrc

patrickelectric, 2024-03-28:
https://gstreamer.freedesktop.org/documentation/rpicamsrc/index.html?gi-language=c

ES-Alexander, 2024-03-28:
I'm pretty sure `rpicamsrc` is [the old/legacy one](https://forums.raspberrypi.com/viewtopic.php?t=326796#p1956315), and what we actually want is support for `libcamerasrc`.

joaoantoniocardoso, 2024-03-28:
Useful links: [[1]](https://indico.freedesktop.org/event/5/contributions/255/attachments/112/170/%5BFINAL%5D%20libcamerasrc_%20Introduction%20and%20usage%20of%20libcamera%27s%20GStreamer%20element.pdf) [[2]](https://libcamera.org/getting-started.html)

MarkSkinner92, 2024-06-24:
What would it take to get this working? I'm running BlueOS on a compute module 4, and it seems my camera does not work with the legacy stack. Having this support would be extremely helpful!