mavlink/mavlink-camera-manager #374: Raspberry Pi libssl.so.1.1: cannot open shared object file: No such file or directory

AlexKlimaj, 2024-04-03:
Trying to run the latest release on a Pi CM4 with the latest raspbian. I was able to get it working with ```sudo apt-get install libssl1.1```

```pi@Pi6X-2:~/Downloads $ ./mavlink-camera-manager-aarch64 
./mavlink-camera-manager-aarch64: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory
pi@Pi6X-2:~/Downloads $ openssl version
OpenSSL 3.0.2 15 Mar 2022 (Library: OpenSSL 3.0.11 19 Sep 2023)
pi@Pi6X-2:~/Downloads $ sudo apt install libssl-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libssl-dev is already the newest version (3.0.11-1~deb12u2+rpt1).
The following packages were automatically installed and are no longer required:
  libmbim-glib4 libmbim-proxy libmbim-utils libqmi-glib5 libqmi-proxy libqmi-utils libqrtr-glib0
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
pi@Pi6X-2:~/Downloads $ ./mavlink-camera-manager-aarch64 
./mavlink-camera-manager-aarch64: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory
pi@Pi6X-2:~/Downloads $ sudo apt-get install libssl1.1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libmbim-glib4 libmbim-proxy libmbim-utils libqmi-glib5 libqmi-proxy libqmi-utils libqrtr-glib0
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  libssl1.1
0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.
Need to get 1,397 kB of archives.
After this operation, 3,841 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.com/debian bookworm/main arm64 libssl1.1 arm64 1.1.1w-0~deb11u1+rpt1 [1,397 kB]
Fetched 1,397 kB in 2s (930 kB/s)    
Preconfiguring packages ...
Selecting previously unselected package libssl1.1:arm64.
(Reading database ... 143399 files and directories currently installed.)
Preparing to unpack .../libssl1.1_1.1.1w-0~deb11u1+rpt1_arm64.deb ...
Unpacking libssl1.1:arm64 (1.1.1w-0~deb11u1+rpt1) ...
Setting up libssl1.1:arm64 (1.1.1w-0~deb11u1+rpt1) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u4) ...
pi@Pi6X-2:~/Downloads $ ./mavlink-camera-manager-aarch64 
2024-04-03T23:48:44.620098Z  INFO main ThreadId(01) src/logger/manager.rs:84: mavlink-camera-manager, version: 0.2.4-2b1dc02, build date: 2024-03-16
2024-04-03T23:48:44.628544Z  INFO main ThreadId(01) src/logger/manager.rs:91: Starting at 2024-04-03T17:48:44
2024-04-03T23:48:44.634540Z  WARN main ThreadId(01) src/settings/manager.rs:128: Failed to load settings file "/home/pi/.config/mavlink-camera-manager/settings.json". Reason: No such file or directory (os error 2)
thread 'main' panicked at src/mavlink/manager.rs:37:51:
No configured mavlink endpoint
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
pi@Pi6X-2:~/Downloads $ 
```

joaoantoniocardoso, 2024-04-04:
Hi @AlexKlimaj, thanks for reporting. I wasn't aware of the `libssl1.1` runtime package dependency, I'll add it to the readme long with the others.

About the last error, it is a known issue, you must run it with the parameter `--reset` on your first time, or whenever you need to change the mavlink endpoint.

joaoantoniocardoso, 2024-04-11:
The need for `--reset` on the first use was fixed by #370, so I'm closing this issue.

Let me know if there is anything else bothering you, thanks!

closed 2024-04-11 by joaoantoniocardoso