mavlink/mavlink-camera-manager #375: Unable to use USB cam

2vin2vin, 2024-04-04:
Hi, I have been working on blueos and I have been trying to use a simple usb camera. This video stream doesn't seem to show out in the GCS(Cockpit). SO, I tried to read the cam feed and encoded it to h264 using ffmpeg command. The GCS reads feed from a websocket, so I read the data in TCP and sent out the same data in websocket to the ip:port, that it was expecting(i.e., 192.168.192.94:6021). But the data doesn't seem to show up. When asked in cockpit they asked to look through mavlink-camera-manger rather than trying out to change in the OS. But I am very much new to rust and mavlink-camera-manager. Any help, would be appreciated.

Thank you

joaoantoniocardoso, 2024-04-04:
Hi @2vin2vin, in BlueOS, when you go to the video page, is your device showing there?

2vin2vin, 2024-04-04:
Yes, it is showing only after I add the stream

joaoantoniocardoso, 2024-04-04:
Right, can you send a screenshot of what you see?

joaoantoniocardoso, 2024-12-03:
Closing due to inactivity. Feel free to comment so we can reopen it.

closed 2024-12-03 by joaoantoniocardoso