mavlink/mavlink-camera-manager #408: API Call CORS-blocked

maphstr, 2024-06-11:
When trying to use some of the API-methods:

Access to XMLHttpRequest at 'http://192.168.40.169:6020/streams' from origin 'http://192.168.40.169:8333' has been blocked by CORS policy: Response to preflight request doesn't pass access control check: No 'Access-Control-Allow-Origin' header is present on the requested resource.

Williangalvani, 2024-06-11:
Hi!
try accessing it through /mavlink-camera-manager , then nginx should handle the CORS permissions

maphstr, 2024-06-12:
Yes. That only works on blueOS though.

patrickelectric, 2024-06-12:
That should be disabled/enabled by a= command line argument in MCM as well

Williangalvani, 2024-06-12:
does it make sense at all to limit cors?
since 99.9% of clients will be external, except when you do a reverse proxy as we do in blueos or are talking to it from a backend program

joaoantoniocardoso, 2024-06-12:
@maphstr, can you try the #409? You can download the binary from [here](https://github.com/joaoantoniocardoso/mavlink-camera-manager/actions/runs/9487055615)

maphstr, 2024-06-13:
Thats what I thought as well Willian, #409 works 👍

PR #409 (2024-06-13): Allow all CORS
Closes #408 

Needs testing.