mavlink/mavlink-camera-manager #420: Allow users to change default video stream creation dynamically, not only via command lines

patrickelectric, 2024-06-24:
At the moment we are creating video stream with rtsp by default, we should allow the users to change it somehow. Like RTSP (TCP) only and etc.

Ref: @goasChris

joaoantoniocardoso, 2024-06-24:
Maybe we can turn this command line into a setting, and allow changing this setting via endpoint?

Shouldn't we do it to all settings? .. Some would require a restart though.

joaoantoniocardoso, 2024-06-24:
For Chris' case, I think we should also allow the RTSP protocol (TCP, UDP, HTTP, etc) to be chosen.

goasChris, 2024-06-25:
Just so I am making sure to communicate correctly. I can currently do it via the `/mavlink-camera-manager/` page:
[image]

It was my belief (could be very wrong), that stream creation happened via the mcm API on 6020 (in BlueOS). I've not dug into this too much tho.

My wish would be to add it into the "video streams" page in BlueOS, as it is a much nicer and easier UX (allow for `rtspt` and co to be used). Even *better* imo, would be to have a drop down with "RTSP" selected, where you can pick:

* TCP (Higher latency, better error recovery on poor connections)
* UDP (Lower latency, no error correction)
etc

Definetly more of a nice to have for users. Don't think our users care much, as long the video don't lag, they are happy, but thinking more for the general audience.

[image]

I honestly assumed that if the MCM api accepts `rtspt` etc, then it would only require a few UI/UX changes to blueOS, but again, I must admit I am not well versed enough in the codebase yes, and I've not thoroughly checked.

joaoantoniocardoso, 2024-06-25:
@goasChris hmm, right, that was done yesterday on BlueOS, already on today's master: https://github.com/bluerobotics/BlueOS/pull/2741

patrickelectric, 2024-06-25:
Soon in beta.6 released today