mavlink/mavlink-camera-manager #436: Support local recording

patrickelectric, 2024-09-02:
Something like the following should be implemented in the camera manager to support local video recording

- [ ] It should support local video recording
    - It should record as a single file or breaking it in 10s
- [ ] Record can be done using MAVLink (MAV_CMD_VIDEO_START_CAPTURE) and via REST API
- [ ] Recording status should be exposed via MAVLink (CAMERA_CAPTURE_STATUS) and via REST API
- [ ] MCM should provide a component id for itself, not the streams, where parameter could be used to control things like mission names
- [ ] Files/Chunks should be store with incremental names \d+.mp4.part (?) inside a "temporary" folder (not true temporary, but for the process) until is complete
- [ ] The folder should be named using an uuid4
- [ ] The folder should move to a finished location once the record is done
- [ ] With the folder, a metadata.json (?) file should be used to store information about the recorded video and session
    - [ ] Camera component ID (for order/serialization)
    - [ ] Mission name (based on `PARAM_EXT_VALUE`)
    - [ ] Finished status (Done [boolean])
- [ ] MCM should provide a frontend and a rest api where it would be possible to:
    - [ ] Start recording
    - [ ] List available records
    - [ ] Download videos
        - [ ] Merge chunks before download  
It would be nicer if:
- [ ] CAMERA_CAPTURE_STATUS includes a stream ID similar to MAV_CMD_VIDEO_START_CAPTURE
- [ ] MAV_CMD_VIDEO_START_CAPTURE includes a human-readable name
- [ ] Component ID (130) is reserved to a camera manager

joaoantoniocardoso, 2025-04-30:
@abhishek202206 I just made the current work visible as a draft PR [here](https://github.com/mavlink/mavlink-camera-manager/pull/484), but it is **not** in a working state yet.

joaoantoniocardoso, 2025-04-30:
Feel free to open a PR with your findings, thanks!