mavlink/mavlink-camera-manager #438: bug: QGC stream switching no longer working - suspected MAVLink issue

ES-Alexander, 2024-09-10:
Per bluerobotics/BlueOS#2892, MCM `t3.12.1` had support for switching between the available camera streams in QGC[^1], which seems to no longer work as of at least MCM `t3.12.4` (possibly earlier).

I haven't looked into QGC or MCM logs, but I imagine this is to do with how MCM lists the available streams over MAVLink, since if QGC isn't receiving that list in a format it understands then it understandably can't make use of the information.

---

MCM was tested within BlueOS (on a Raspberry Pi 3B), so there's some possibility of it just being a MAVLink routing issue, although that seems unlikely.

Main receiver testing was in QGC 4.3.0, on macOS Sonoma 14.5.

[^1]:with a joystick button or with a dropdown available from the cog icon at the right side of the recording widget

joaoantoniocardoso, 2024-12-10:
This is still a mystery. I can see the QGC toggle messages, but not a single hint is shown in the console/logs.

[image]

I tested the video toggle via joystick with QGC 4.1.2 and 4.3 with MCM [t3.12.1](https://github.com/mavlink/mavlink-camera-manager/releases/tag/t3.12.1), [t3.12.0](https://github.com/mavlink/mavlink-camera-manager/releases/tag/t3.12.0), [t3.9.0](https://github.com/mavlink/mavlink-camera-manager/releases/tag/t3.9.0), and [t3.4.0](https://github.com/mavlink/mavlink-camera-manager/releases/tag/t3.4.0) without success.

When switching the camera from the Dropdown menu, all versions worked.

joaoantoniocardoso, 2024-12-10:
Okay, when looking at QGC's source code, I realized that when testing, I was using the toggle video source action, but there are others.

- The next/previous camera action works
- The next/previous stream action doesn't work.

Tested on t3.17.0, t3.12.1, and t3.4.0 but only with QGC 4.3, since 4.1 doesn't seem to have those actions.

Maybe there is something else going on?