mavlink/mavlink-camera-manager #445: Question: How to test that mavlink works?

phuesler, 2024-10-22:
Hi,

Thank you for creating and contributing to this useful project.

I'm struggling to understand the flow of data and hope that you can shed some light on it. Where can I test that I can start/top etc. the camera through mavlink commands? I was hoping this would work with QGrouncControl.

 As I understand it, there are 4 components involved:

* A drone running a flight controller that speaks Mavlink. Or a sitl like ArduPilot sim_vehicle.py
* mavproxy
* mavlink-camera-manager
* A GCS like QGroundControl

Which components are connected together? I assumed that QGroundControl and mavlink-camera-manager, there is no need to involve the flight controller. But I'm starting to think that his is wrong. Because I can't get it to work:

I followed the example in the Readme and issued the following commands:

```
./Tools/autotest/sim_vehicle.py -v ArduCopter
```

```
./mavlink-camera-manager-linux-desktop --verbose --mavlink-system-id 1 --mavlink tcpout:0.0.0.0:5760
```

```
./QGroundControl.AppImage
```

I then changed the QGroundControl Application settings to listen on port 5600 for a UDP stream. Once configured in the UI of mavlink-camera-manager, the video shows up in QGroundControl.

When I check the output of mavlink-camera-manager, it doesn't say anything about a successful mavlink connection. It is outputting the same debug message about mavlink HEARTBEAT, regardless of weather QGroundControl is running or not.

What am I missing here?

joaoantoniocardoso, 2024-10-22:
Hi @phuesler,

Yeah, the exactly thing you are missing is to ask for mavproxy for an exclusive endpoint for mavlink camera manager.

```
./Tools/autotest/sim_vehicle.py -v ArduCopter --mavproxy-args --out=tcpin:0.0.0.0:5777
```

and then:

```
./mavlink-camera-manager-linux-desktop --verbose --mavlink-system-id 1 --mavlink tcpout:0.0.0.0:5777
```

I hope this helps,
thanks!

phuesler, 2024-10-23:
Thanks for your quick reply and for taking the time.

Unfortunately I see still the same behavior, issuing the above commands.

* mavlink-camera-manager is sending mavlink messages to mavproxy who sends it to the autopilot? 
* mavlink-camera-manager is broadcasting its mavlink heartbeats announcing its camera?
* Does QGroundControl send the camera related messages to ArduPilot and not mavlink-camera-manager directly?
* What is the expected behavior in QGroundControl when it 'works'?

If it helps, I see this in the SITL console:

```
COMMAND_ACK: REQUEST_CAMERA_INFORMATION: UNSUPPORTED
```

joaoantoniocardoso, 2024-12-06:
Hi @phuesler,

sorry for my late reply, I hope you already got it working. In any case, here is my answer for your questions:

> mavlink-camera-manager is sending mavlink messages to mavproxy who sends it to the autopilot?
Correct. You can also use [Mavlink Server](https://github.com/bluerobotics/mavlink-server/) as an alternative router:

```bash
mavlink-server tcpclient:127.0.0.1:5760 udpserver:127.0.0.1:14550 tcpserver:127.0.0.1:5777 --verbose
```

> mavlink-camera-manager is broadcasting its mavlink heartbeats announcing its camera?

Correct, as long as you don't check the `Disable Mavlink` box when configuring your stream.

> Does QGroundControl send the camera related messages to ArduPilot and not mavlink-camera-manager directly?

Correct, any mavlink client sends data to someone who routes them to the target or broadcasts it to everybody.

> What is the expected behavior in QGroundControl when it 'works'?

When it works, it should show `Connected` on the topbar, and any stream configured on MCM should be available. 

I just ran a localhost test here, and for QGC to communicate just fine:

https://github.com/user-attachments/assets/20e5f7fe-982a-433d-a00c-8666bfaa8123

Let me know if you need more help, have a great week!

joaoantoniocardoso, 2024-12-06:
oops, I just noticed that my test was using a stream configured on `udp://127.0.0.1:5600`, and QGC was getting it automatically without using the Mavlink protocol. When I changed it to `udp://127.0.0.1:5601`, it received no streams. 

I'll do some more testing here, but my first suspect is that maybe the autopilot isn't doing the routing.

joaoantoniocardoso, 2024-12-06:
well, that's definitely a router configuration problem. I'm not very familiar with mavproxy, but this worked:
```bash
Tools/autotest/sim_vehicle.py -v ArduCopter --mavproxy-args '--master=tcp:0.0.0.0:5760 --master=tcpin:0.0.0.0:5777 --master=udpin:0.0.0.0:14550'
```

```bash
mavlink-camera-manager --verbose --mavlink tcpin:127.0.0.1:5777
```

phuesler, 2024-12-06:
Hi @joaoantoniocardoso,

Thank you very much for taking the time. Now it indeed seems to work. ArduPilot shows that Links with new components and QGroundControl let's me inspect the camera's settings, which wasn't the case before. I used the commands above as with the latest camera manager release and the latest qgroundcontrol release.

I will run some more tests with missions, but so far so good. Once again, thank you. I appreciate you taking the time.

joaoantoniocardoso, 2024-12-06:
Good to know!

Btw, I don't know what your project is about, but maybe you could benefit from using [BlueOS](https://github.com/bluerobotics/BlueOS) and [Cockpit](https://github.com/bluerobotics/Cockpit), which has [Mavlink Server](https://github.com/bluerobotics/mavlink-server) and other handy tools.

I'll close this for now, we can reopen if needed, so let me know if you need more help.

closed 2024-12-06 by joaoantoniocardoso