mavlink/mavlink-camera-manager #460: Redirect stream: some cameras require `force-non-compliant-url=true` in order to work

Williangalvani, 2025-01-18:
Gstreamer fails to negotiate without it on the rtspsrc. ffmpeg and vlc can happily play the stream. I built a version.

[this](https://github.com/mavlink/mavlink-camera-manager/compare/master...Williangalvani:mavlink-camera-manager:master?expand=1) was enough for getting the camera stream to work, but I'm afraid it could break other cameras, so perhaps we should instead update `get_encode_from_stream_uri()` so it tries both ways?

joaoantoniocardoso, 2025-02-22:
Right, based on [this](https://gitlab.freedesktop.org/gstreamer/gst-plugins-good/-/issues/922) and [this](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/1447) (that mainly leads to [this](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/1447)), it is probably failing to detect that it should do that as the fallback, so.. I guess we should try to do it by ourselves as you suggested.

joaoantoniocardoso, 2025-05-16:
The solution to be implemented should be to implement a strategy (pattern) there, using the default to begin with (`force-non-compliant-url=false`), then falling back to `force-non-compliant-url=true`.