mavlink/mavlink-camera-manager #462: bug: build.rs calls reqwest directly, ignoring proxy settings from env

mokurin000, 2025-02-07:


joaoantoniocardoso, 2025-02-22:
Hi!

Are you using the `HTTPS_PROXY` env var? Is it a socket?

joaoantoniocardoso, 2025-02-22:
> > Hi!
> > Are you using the `HTTPS_PROXY` env var? Is it a socket?
> 
> Yeah, I set it to a http proxy `http://127.0.0.1:7897` and that was ignored
> 
> Strange but I found `reqwest::blocking::get()` will create a default Client, which will read system proxies by default
> 
> ```rust
> Client::builder().build()?.get(url).send()
> ```

Yeah, but maybe it needs to be an https proxy, and not http?

mokurin000, 2025-02-23:
> > > Hi!
> > > Are you using the `HTTPS_PROXY` env var? Is it a socket?
> > 
> > 
> > Yeah, I set it to a http proxy `http://127.0.0.1:7897` and that was ignored
> > Strange but I found `reqwest::blocking::get()` will create a default Client, which will read system proxies by default
> > Client::builder().build()?.get(url).send()
> 
> Yeah, but maybe it needs to be an https proxy, and not http?

I found the reason

```rust
fn get_from_environment() -> SystemProxyMap {
    let mut proxies = HashMap::new();

    if !(insert_from_env(&mut proxies, "http", "ALL_PROXY")
        && insert_from_env(&mut proxies, "https", "ALL_PROXY"))
    {
        insert_from_env(&mut proxies, "http", "all_proxy");
        insert_from_env(&mut proxies, "https", "all_proxy");
    }

    if is_cgi() {
        if log::log_enabled!(log::Level::Warn) && env::var_os("HTTP_PROXY").is_some() {
            log::warn!("HTTP_PROXY environment variable ignored in CGI");
        }
    } else if !insert_from_env(&mut proxies, "http", "HTTP_PROXY") {
        insert_from_env(&mut proxies, "http", "http_proxy");
    }

    if !insert_from_env(&mut proxies, "https", "HTTPS_PROXY") {
        insert_from_env(&mut proxies, "https", "https_proxy");
    }

    proxies
}
```

https://docs.rs/reqwest/0.12.12/src/reqwest/proxy.rs.html#914

Without `HTTPS_PROXY`, https traffics will not be proxied by default

closed 2025-02-23 by mokurin000

joaoantoniocardoso, 2025-02-23:
Thanks for letting me know!