mavlink/mavlink-camera-manager #464: Stream missing from Available Producers

Meekohi, 2025-02-28:
The only options I see for my video source Encode are MJPG and YUVV. In contrast for the fake source I see h264, h265, MJPG, and YUVV.
* h264 works as expected 👍🏼 
* h265 says it is incompatible with UDP
* The others allow me to create a stream, but nothing appears in "Available Producers" on the WebRTC Development UI.

Is there a way to either make h264 available for my video stream, or advice on how to figure out what is wrong with the mjpg/yuvv options?

joaoantoniocardoso, 2025-02-28:
Oh, also, be aware that stream encoding requires some CPU or GPU process power. Feel free to try different encoders than the `x264enc` that I used in the example there.

ben-xD, 2025-03-10:
If i want to use WebRTC with a browser (or electron app) than supports H265 webrtc tracks, does mavlink-camera-manager support that?

ben-xD, 2025-03-10:
Thanks a lot for your response. Was about to stop hacking but your help gave me extra willpower 😅️! MCM is fast 😃️

I've got fake H265 video generated from the MCM web UI streaming into Chrome, though I have observed the same effect with MCM and my own Python (gstreamer webrtc) implementation: video is extremely laggy / frozen (0.01FPS at best). Not sure why but definitely not an MCM issue.

However, when trying redirect my own video, I see in the server backend logs
```
Scheme "udp265" is not supported for Redirect Pipelines
```

I am unable to delete this stream Is there a way to clear the state?

[image]

---

## Details only useful for people trying to emulate my setup

I've started Chrome with H265 support on macOS and visited webrtc dev UI `http://127.0.0.1:6020/webrtc/index.html`:
```
/Applications/Google\ Chrome.app/Contents/MacOS/Google\ Chrome --remote-debugging-port=9222  --enable-features=PlatformHEVCEncoderSupport,WebRtcAllowH265Receive,WebRtcAllowH265Send --force-fieldtrials=WebRTC-Video-H26xPacketBuffer/Enabled
```

I'm using the MCM web UI with the following config:
[image]

I was getting back an error on macOS with the fake video stream because it was using `x265enc`, I swapped it out with `vtenc_h265`.
```
Error {\n    domain: gst_parse_error,\n    code: 1,\n    message: \"no element \\\"x265enc\\\"\",\n}. Trying again in one second...
```

[image]

I'm also emulating a camera I have with gstreamer on macOS and now trying to use the "Camera: Redirect source" section of the MCM Web UI:
```bash
gst-launch-1.0 videotestsrc ! video/x-raw,width=1280,height=720,framerate=30/1 ! vtenc_h265 ! h265parse ! rtph265pay config-interval=1 ! udpsink host=127.0.0.1 port=11035
```

joaoantoniocardoso, 2025-03-10:
Nice setup, I didn't know how to run Chrome with H265 WebRTC support!

> Scheme "udp265" is not supported for Redirect Pipelines

The `udp265` is in fact a confusing thing: it has historical reasons to allow QGC to know that UDP is a h265 and not a H264. The actual stream lives at `udp://`.

> error_message="Failed getting CaptureConfiguration from endpoint, trying again soon..."

This means that our internal `udpsrc` is failing to fetch frames from that address, could you run it with `--verbose` and check the logs for more information?

The current behavior (merged today) is that when a stream is added, there's a task in the background that keeps forever-trying to run the stream.

> Solution: Just delete rm -rf ~/.config/mavlink-camera-manager/settings.json

Or run it with `--reset` arg.

> video in Chrome is extremely laggy

About the lag, could you check if (1) the same lag happens with (the latest) Safari, and if (2) the lag isn't in the encoding side? Maybe receive the video using something like:

```bash
gst-launch-1.0 -vc udpsrc address=127.0.0.1 port=11035 close-socket=false do-timestamp=true \
    ! application/x-rtp \
    ! rtph265depay \
    ! decodebin3 \
    ! videoconvert ! autovideosink sync=false
```

---

BTW, your UDP stream might be better if you use `sync=false` in your `udpsink`.

---

Here's an example of my h265 test pipeline on Linux:

Sender:
```bash
gst-launch-1.0 -vc videotestsrc pattern=ball \
    ! video/x-raw,format=UYVY,width=320,height=240,framerate=30/1 \
    ! videoconvert ! x265enc bitrate=5000 tune=4 \
    ! h265parse ! queue ! rtph265pay name=pay0 config-interval=10 pt=96 \
    ! multiudpsink clients=192.168.2.1:5609
```

Receiver:
```bash
gst-launch-1.0 -vc udpsrc address=192.168.2.1 port=5608 close-socket=false auto-multicast=true \
    ! application/x-rtp, payload=96 \
    ! rtph265depay \
    ! avdec_h265 \
    ! videoconvert ! autovideosink sync=false
```

Thanks!

ben-xD, 2025-03-11:
Thanks, I"ll use `--reset` and `sync=false` :). Here's the verbose log and safari test / gstreamer playback results. This has been thrilling but it is 12AM now, I'll be back tomorrow. Thanks for your help. 
---

## Verbose MCM logging
After creating a redirect source with `udp://127.0.0.1:11035` in the web UI, 

<details>
  <summary>Click to expand</summary>
```bash
./mavlink-camera-manager --reset --verbose
2025-03-11T00:02:43.987367Z  INFO src/lib/logger/manager.rs:193: mavlink-camera-manager, version: 0.2.4-9fbc575, build date: 2025-03-10
2025-03-11T00:02:43.987492Z  INFO src/lib/logger/manager.rs:200: Starting at 2025-03-11T00:02:43
2025-03-11T00:02:43.987511Z  INFO src/lib/logger/manager.rs:204: Server running at 0.0.0.0:6020
2025-03-11T00:02:43.987522Z DEBUG src/lib/logger/manager.rs:205: Command line call: ./mavlink-camera-manager --reset --verbose
2025-03-11T00:02:43.987536Z DEBUG src/lib/logger/manager.rs:206: Command line input struct call: Args {
    mavlink: "udpin:0.0.0.0:14550",
    settings_file: "~/.config/mavlink-camera-manager/settings.json",
    default_settings: None,
    reset: true,
    rest_server: "0.0.0.0:6020",
    stun_server: "stun://0.0.0.0:3478",
    turn_servers: [],
    signalling_server: "ws://0.0.0.0:6021",
    verbose: true,
    gst_feature_rank: [],
    log_path: Some(
        "./logs",
    ),
    enable_tracing_level_log_file: false,
    vehicle_ddns: None,
    enable_tracy: false,
    enable_thread_counter: false,
    enable_webrtc_task_test: None,
    mavlink_system_id: 1,
    onvif_auth: [],
}
2025-03-11T00:02:43.987547Z DEBUG src/lib/settings/manager.rs:78: Settings reset, an empty settings will be loaded and stored as "/Users/user/.config/mavlink-camera-manager/settings.json".
2025-03-11T00:02:43.987572Z  INFO src/lib/settings/manager.rs:116: The settings file "/Users/user/.config/mavlink-camera-manager/settings.json" will be backed-up as "/Users/user/.config/mavlink-camera-manager/settings.json.bak", and a new (empty) settings file will be created in its place.
2025-03-11T00:02:43.988461Z DEBUG init: src/lib/stream/manager.rs:64: Starting video stream service.
2025-03-11T00:02:43.996857Z DEBUG src/lib/settings/manager.rs:175: Settings saved: ManagerStruct {
    file_name: "/Users/user/.config/mavlink-camera-manager/settings.json",
    config: SettingsStruct {
        header: HeaderSettingsFile {
            name: "Camera Manager",
            version: 0,
        },
        mavlink_endpoint: "udpin:0.0.0.0:14550",
        streams: [],
    },
}
2025-03-11T00:02:43.996911Z DEBUG default: src/lib/stream/webrtc/signalling_server.rs:48: Starting SignallingServer task...
2025-03-11T00:02:43.996960Z DEBUG start_default: src/lib/stream/manager.rs:130: Streams: []
2025-03-11T00:02:43.996962Z DEBUG src/lib/stream/webrtc/signalling_server.rs:51: SignallingServer task started!
2025-03-11T00:02:43.997049Z DEBUG runner{endpoint=Url { scheme: "ws", cannot_be_a_base: false, username: "", password: None, host: Some(Ipv4(0.0.0.0)), port: Some(6021), path: "/", query: None, fragment: None }}: src/lib/stream/webrtc/signalling_server.rs:76: Signalling server: listening on: 0.0.0.0:6021
2025-03-11T00:02:43.997055Z  INFO /Users/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/actix-server-2.3.0/src/builder.rs:240: starting 11 workers
2025-03-11T00:02:43.997075Z  INFO /Users/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/actix-server-2.3.0/src/server.rs:197: Tokio runtime found; starting in existing Tokio runtime
2025-03-11T00:02:44.993342Z DEBUG receiver_loop:connect{address="udpin:0.0.0.0:14550"}: src/lib/mavlink/manager.rs:235: Connecting...
2025-03-11T00:02:44.993509Z  INFO receiver_loop:connect{address="udpin:0.0.0.0:14550"}: src/lib/mavlink/manager.rs:239: Successfully connected
2025-03-11T00:03:57.313811Z DEBUG HTTP request{http.method=POST http.route=/streams http.flavor=1.1 http.scheme=http http.host=0.0.0.0:6020 http.client_ip=127.0.0.1 http.user_agent=Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/134.0.0.0 Safari/537.36 http.target=/streams otel.name=HTTP POST /streams otel.kind="server" request_id=696bc138-1e0d-43f7-ba99-545d9fbaaa91}:add_stream_and_start{video_and_stream_information=VideoAndStreamInformation { name: "Redirect", stream_information: StreamInformation { endpoints: [Url { scheme: "udp", cannot_be_a_base: false, username: "", password: None, host: Some(Domain("127.0.0.1")), port: Some(11035), path: "", query: None, fragment: None }], configuration: Redirect(RedirectCaptureConfiguration), extended_configuration: Some(ExtendedConfiguration { thermal: false, disable_mavlink: true }) }, video_source: Redirect(VideoSourceRedirect { name: "Redirect source", source: Redirect("Redirect") }) }}:try_new{video_and_stream_information=VideoAndStreamInformation { name: "Redirect", stream_information: StreamInformation { endpoints: [Url { scheme: "udp", cannot_be_a_base: false, username: "", password: None, host: Some(Domain("127.0.0.1")), port: Some(11035), path: "", query: None, fragment: None }], configuration: Redirect(RedirectCaptureConfiguration), extended_configuration: Some(ExtendedConfiguration { thermal: false, disable_mavlink: true }) }, video_source: Redirect(VideoSourceRedirect { name: "Redirect source", source: Redirect("Redirect") }) }}: src/lib/stream/mod.rs:92: Starting StreamWatcher task...
[... cut ...]
https://github.com/user-attachments/assets/0337b0bb-c266-48a6-b2d6-bf0c27712817

Oh yea, I definitely tested watching the video with gstreamer, and it plays fine
- video output: `gst-launch-1.0 videotestsrc ! video/x-raw,width=1280,height=720,framerate=30/1 ! vtenc_h265 ! h265parse ! rtph265pay config-interval=1 ! udpsink host=127.0.0.1 port=11035 sync=false`
- video input: `gst-launch-1.0 -v udpsrc port=11035 caps="application/x-rtp, media=video, encoding-name=H265, payload=96" ! rtph265depay ! avdec_h265 ! videoconvert ! autovideosink sync=false`

[image]

joaoantoniocardoso, 2025-03-11:
Right, I was able to reproduce the problem here.

I am using the `typefind` element to check the received encode, but it actually only works with RTSP because RTSP provides that via SDP. UDP doesn't implement anything like that, so we'll have to think of another strategy for it.

Now it makes sense the workaround that QGC did: if it's `udp://`, it considers an RTP H264-encoded stream, if it's `udp265://`, then it considers it's an RTP H265-encoded stream.

I'll rework that with a more robust approach, for now, it only works with redirect RTSP streams.

I'll let you know here when it's ready, thanks!

ben-xD, 2025-03-11:
Nice, that works now (though with same freezing issue in Chrome). Thanks a lot.
[image]

## Minor issue

I did notice a bug with deleting redirected sources, after successfully creating it:

[image]

joaoantoniocardoso, 2025-03-12:
> Nice, that works now (though with same freezing issue in Chrome). Thanks a lot.

Awesome! That'll soon be on master =)

> I did notice a bug with deleting redirected sources, after successfully creating it:

Thanks for reporting it, I'll check this asap!

closed 2025-04-02 by Meekohi