mavlink/mavlink-camera-manager #478: H264 Encoding error when running UVC Camera

abhishek202206, 2025-03-27:
I am trying to run UVC camera ARDUCAM IMX323 using mavlink-camera-manager but when I try to stream H264 encoding the gstreamer fails with the following error.

```
Name: /dev/video4 - USB 2.0 Camera: Arducam IMX323
Video: H264 640x480 @ 30 / 1 FPS, Thermal: falseDisable Mavlink: false

Delete stream
Endpoints:

udp://192.168.0.247:5600

Configuration:

{
  "id": "97cd52da-8fd5-4072-bc55-5ebd46c590a8",
  "running": false,
  "error": "Failed to recreate the stream 97cd52da-8fd5-4072-bc55-5ebd46c590a8: Error {\n    domain: gst_parse_error,\n    code: 2,\n    message: \"no property \\\"aggregate-mode\\\" in element \\\"rtph264pay22\\\"\",\n}. Trying again in one second...",
  "video_and_stream": {
    "name": "/dev/video4 - USB 2.0 Camera: Arducam IMX323 ",
    "stream_information": {
      "endpoints": [
        "udp://192.168.0.247:5600"
      ],
      "configuration": {
        "type": "video",
        "encode": "H264",
        "height": 480,
        "width": 640,
        "frame_interval": {
          "numerator": 1,
          "denominator": 30
        }
      },
      "extended_configuration": {
        "thermal": false,
        "disable_mavlink": false
      }
    },
    "video_source": {
      "Local": {
        "name": "USB 2.0 Camera: Arducam IMX323 ",
        "device_path": "/dev/video4",
        "type": {
          "Unknown": "usb-xhci-hcd.2.auto-1"
        }
      }
    }
  }
}
```

If I stream the data in MJPG format then the gstreamer pipeline works but gives an error while running H264 encoding.

I am getting the mavlink connection via QGC correctly.

The application was built on arm64 architecture

Below is the screenshot of webpage running on my embedded device

[image]

joaoantoniocardoso, 2025-03-27:
Hi, that's probably the gstreamer version.

Could you try at least 1.24? Thanks!

abhishek202206, 2025-03-28:
Hi I don't think 1.24 would work on my chipset. As I using msm-5.4 kernel with Ubuntu 20.04. So that is the reason I am unable to update. 

are there any special steps I need to do for updating the gstreamer version for my HW.

abhishek202206, 2025-03-31:
Thanks @joaoantoniocardoso . I will try to build or update gstreamer version within the docker build. hopefully that would work. Will keep you posted in case I get stuck. Thanks for you quick response and support

abhishek202206, 2025-04-01:
So I tried to remove the aggregate mode from the code just to make sure if I can get the stream but now I am getting the error that imagesink is unable to find avdec_h264 even though I have the  package build and available

```
exalinux@qrb5165-rb5:~/mav_cam_updated/mavlink-camera-manager/target/release$ gst-inspect-1.0 | grep avdec_h264
libav:  avdec_h264: libav H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 decoder

```

[image]

I am able to run gstreamer imagesink using avdec_h264 from terminal but not from within the application

joaoantoniocardoso, 2025-04-01:
Hi! Could you check if the `lowres` property is available in your `avdec_h264` plugin?

```bash
gst-inspect-1.0 avdec_h264 | grep lowres
```

joaoantoniocardoso, 2025-04-01:
hmm, try to run the following command before invoking MCM:
```
export GST_DEBUG='*:3;*avdec*:4;'
```

joaoantoniocardoso, 2025-04-01:
See if anything weird pops out. Also, try testing the decoder with:
```
gst-launch-1.0 -vc videotestsrc ! x264enc ! h264parse ! avdec_h264 ! videoconvert ! autovideosink
```

(use any other encoder if you don't have x264enc)

abhishek202206, 2025-04-01:
Here is the output
```
exalinux@qrb5165-rb5:~/mav_cam_updated/mavlink-camera-manager/target/release$ gst-launch-1.0 -vc videotestsrc ! x264enc ! h264parse ! avdec_h264 ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw, width=(int)320, height=(int)240, framerate=(fraction)30/1, format=(string)Y444, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
Redistribute latency...
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:sink: caps = video/x-raw, width=(int)320, height=(int)240, framerate=(fraction)30/1, format=(string)Y444, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:src: caps = video/x-h264, codec_data=(buffer)01f4000dffe1001d67f4000d919b28283f602d41804150000003001000000303c8f142996001000568ebec4480, stream-format=(string)avc, alignment=(string)au, level=(string)1.3, profile=(string)high-4:4:4, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, codec_data=(buffer)01f4000dffe1001d67f4000d919b28283f602d41804150000003001000000303c8f142996001000568ebec4480, stream-format=(string)avc, alignment=(string)au, level=(string)1.3, profile=(string)high-4:4:4, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, chroma-format=(string)4:4:4, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
Redistribute latency...
/GstPipeline:pipeline0/avdec_h264:avdec_h264-0.GstPad:sink: caps = video/x-h264, codec_data=(buffer)01f4000dffe1001d67f4000d919b28283f602d41804150000003001000000303c8f142996001000568ebec4480, stream-format=(string)avc, alignment=(string)au, level=(string)1.3, profile=(string)high-4:4:4, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, chroma-format=(string)4:4:4, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = video/x-h264, codec_data=(buffer)01f4000dffe1001d67f4000d919b28283f602d41804150000003001000000303c8f142996001000568ebec4480, stream-format=(string)avc, alignment=(string)au, level=(string)1.3, profile=(string)high-4:4:4, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/avdec_h264:avdec_h264-0.GstPad:src: caps = video/x-raw, format=(string)Y444, width=(int)320, height=(int)240, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt601, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, width=(int)320, height=(int)240, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, format=(string)YV12
/GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw, width=(int)320, height=(int)240, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, format=(string)YV12
/GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0/GstXvImageSink:autovideosink0-actual-sink-xvimage.GstPad:sink: caps = video/x-raw, width=(int)320, height=(int)240, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, format=(string)YV12
[... cut ...]
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:sink: caps = video/x-raw, format=(string)Y444, width=(int)320, height=(int)240, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)bt601, framerate=(fraction)30/1

```

Nothing weird pops out I can see the test screen

abhishek202206, 2025-04-01:
I think so I got the error. It is failing to load libgstlibav.so

```
(mavlink-camera-manager:101178): GStreamer-WARNING **: 14:19:36.448: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlibav.so': /lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block

```

joaoantoniocardoso, 2025-04-01:
I've never seen that, maybe that's a limitation from the OS or how the library was built?

closed 2025-04-01 by abhishek202206

joaoantoniocardoso, 2025-04-01:
The 'code' that you are referring to is your own application?

I've never used qtic2venc. Overall the `gst-inspect-1.0 qtic2venc` should tell you what caps it accepts as input and output, then you might need to put a capsfilter before and/or after. Sometimes we need to add some other element to convert the format before the encoder, like a `videoconvert`.

If you tell me what kind of problem you are having I can give you more specific tips

abhishek202206, 2025-04-01:
Yes the code I am referring to is for my own application but what I meant was if someone who wants to use custom HW encoders can use the branch as reference. I can see that there was a request as enhancement for using custom features so I thought I could contribute.

I will start building using custom HW encoder on my hardware and will share provide more detailed error with logs.

Thanks for your help. we can close this task now

joaoantoniocardoso, 2025-04-01:
Great, yeah, we have a plan for an architecture change to accommodate encoders and decoders here in MCM, though we haven't invested in that yet.

abhishek202206, 2025-04-01:
TBH I would love to contribute to the enhancement you have proposed and other functionality to the MCM. Let mw know how can I contribute and help

abhishek202206, 2025-04-01:
Sure No problem by the time I will test locally and hardcode the encoders to see what error comes and once you create a branch and I will start supporting